adafruit/Raspberry-Pi-Installer-Scripts #185: i2s audio not working for raspberry os 32 bit 2021-05-07

victorcid, 2021-08-12:
i tried to install the module max98357a  butt it does not sound any of the tests
after instalation it shows when i use alsamixer it shous as a divice but i cant findit on the menu for sound divices also
whrn i use aplay -l it says
card 0: sndrpihifiberry [sdn_rpi_hifiberry_dac], device 0: hifiiberry DAC Hifi pc m5101a-hifi-0 
  subdevices: 0/1
  subdevice #0: subdevice #0

ive alredy try the manual instalation and updating drivers

makermelissa, 2021-08-12:
Just to be clear, which script are you running? `i2samp.sh` or something else?

makermelissa, 2021-09-10:
No response, but learn guide recommends `i2samp.sh` for that amp, so I'm going with that.

newhinton, 2022-10-20:
I could also not get this working.

I wanted to use this dac for retropie with the pitft, but even removing the pitft didn't help.

I used the script to install everything (after triple-checking the wiring), but the test-step fails:

```
Do you wish to test your system now? [y/N] y
Testing...

speaker-test 1.1.8

Playback device is default
Stream parameters are 48000Hz, S16_LE, 2 channels
WAV file(s)
Playback open error: -2,No such file or directory
```

```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 [HifiBerry DAC HiFi pcm5102a-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

```
$ alsamixer
cannot open mixer: No such file or directory

```
```
$ alsamixer -c 0
Card: snd_rpi_hifiberry_dac
Chip:
...
```
etc. Chip is actually empty.

Any idea how i can fix this?

newhinton, 2022-10-24:
@makermelissa It seems that this is an issue with the current Raspbian.
Even a fresh install, on Raspbian Lite (32Bit) does not result in the i2s amplifier to output sound.
`Linux raspberrypi 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l GNU/Linux`

I have confirmed that the amplifier works with a pico. And a second board also did not work.

Though i could get `Subdevices: 0/1` to be `Subdevices: 1/1`, i had a broken `.asound.conf`-file

newhinton, 2022-10-27:
Okay i found the culprit:

retrogame

The controller application grabs the required pins from the adc, and therefore the dac does not work.

makermelissa, 2022-10-27:
Thanks. Since you're using a PiGRRL 2.0 Custom PCB, I presume you've been following the [PiGRRL Guide](https://learn.adafruit.com/pigrrl-2). In there, it appears there is a page with a section on remapping the buttons: https://learn.adafruit.com/pigrrl-2/software#remapping-controls-2141813. Do you think this would have worked?

Also, can you confirm that you were running `retrogame.sh`? Thanks.

newhinton, 2022-10-27:
Yes i have been following the guide.

In the end i remapped the buttons used by retrogame, the tool suggested by the guide.

However, before i could use different buttons, i had to physically modify the pcb. After doing so, remapping the buttons worked!

For anyone interested, the pins used by the DAC are 18, 19 and 21.

It seems that the default pigrrl2 configuration has gpio1, 2, 3 and 21 unused, all of the others are in use by either the gamepad or the screen.

makermelissa, 2022-10-27:
Thanks. I'll leave this open for now and see if there's a cleaner way to resolve this that doesn't involve modifying the board.

makermelissa, 2022-11-03:
After looking at this more carefully, the guide has you wire up a PAM8302 by soldering directly to the headphone jack terminals. As mentioned, the max98357a and default pigrrl2 configuration are pin incompatible, so I think @newhinton came up with a great solution.

As for the OP, it was likely a software conflict, but I can't know for certain without more information. Closing since there's nothing more to do on this issue.

closed 2022-11-03 by makermelissa