adafruit/Raspberry-Pi-Installer-Scripts #187: Braincraft hat not working with latest RPi kernel

jerryneedell, 2021-08-24:
I tried installing the pitft display for the Braincraft hat following the guide 
https://learn.adafruit.com/adafruit-braincraft-hat-easy-machine-learning-for-raspberry-pi

The install went fine, but when I tried to test it with
raspistill -t 0
I get nothing on the display.

It had been working prior to updating the Raspberry Pi kernel

I also noted that with the latest kernel, I do not see anything at all on the deeply, no console messages like I usually do, except when I reboot, there is a momentary flash of the linux console on the screen as it reboots but nothing is there one it finishes the reboot and I can't get anything to display.
i do not get any errors, just no display.

I tried doing a full clean install of RaspiOS and followed the guide but get the same results.

I did follow the troubleshooting instructions and it does work if I revert the kernel back to 5.4.79 per the guide.

This does work but I was wondering if a fix was in the works to allow more recent kernel to be used.

makermelissa, 2021-08-24:
Is this happening on a Pi 4 or a different one?

jerryneedell, 2021-08-24:
Yes -- Pi 4 -- sorry I left that out.

jerryneedell, 2021-08-24:
I am working thought some of the newer kernels:
sudo sh rpi-pin-kernel-firmware.sh 1.20210430-2
seems to work

makermelissa, 2021-08-24:
Thank you. Yeah figuring out the latest working kernel may be the answer for now.

jerryneedell, 2021-08-24:
interesting -- I installed
sudo sh rpi-pin-kernel-firmware.sh 1.20210527-1
and it worked until I reinstalled the Audio -- that appears to also re-install the kernel....and broke it
I am re-installing it...

jerryneedell, 2021-08-24:
reinstalling sudo sh rpi-pin-kernel-firmware.sh 1.20210527-1  now works.
and the Audio is still working.
I have to stop for today but will try some newer kernels in the morning -- possible it is the audio script that was messing things up....

makermelissa, 2021-08-24:
Awesome, thanks.

jerryneedell, 2021-08-25:
So -- it appears that 20210527-1  works OK but 20210727-1  fails
Note that I had the above issue when I installed the audio v5.9 per the guide after installing 20210527-1 -- it appeared to update the kernel to 20200805-1 and did not work, but I then reinstalled 20210527-1 and it is now working fine.  
I am confused by this and will do more troubleshooting today, but it is working -- audio as well.

jerryneedell, 2021-08-25:
I have confirmed that with kernel 20210527-1 installed and pinned running the install.sh script from the seed- voicecard folder updates the kernel to 20210805-1 so it no longer works. Reinstalling/pinning  kernel 20201527-1 fixes it.  
sudo sh rpi-pin-kernel-firmware.sh 1.20210527-1

jerryneedell, 2021-08-25:
In the install script  to pin the kernel it looks like it has the 202021126-1 version hard coded when it writes the file to /etc/apt/prefereneces.d <https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/rpi-pin-kernel-firmware.sh#L38> 

so this file is not correct when I run
sudo sh rpi-pin-kernel-firmware.sh 1.20210527-1
It is not clear to me how this file is actually used but ti looks like it should be changed.

jerryneedell, 2021-08-25:
Maybe I am misinterpreting what "pinning" means. Running a sudo apt update;sudo apt full-upgrade  allow installs the newest kernel. I guess you just have to revert the kernel any time it is updated.
I thought the "pinning" prevented apt from upgrading the kernel....

makermelissa, 2021-08-25:
I don't think you're misinterpreting. It should keep an older version when you attempt to update to the latest. Also, the version shouldn't be hardcoded. Good catch.

makermelissa, 2021-09-07:
Ok, after installing the drivers as the instructions, the screen was black. As some other recent issues mentioned, the backlight wasn't turning on for the display, so I opened python and ran this code:
```
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> import digitalio
>>> backlight = digitalio.DigitalInOut(board.D26)
```
The Backlight turned on and the correct info is all there, so it looks like the issue is just related to the backlight not turning on by default. This should be a fairly simple fix of modifying the device tree overlays.

jerryneedell, 2021-09-07:
Nice! Is this with the latest RPi kernel installed?

makermelissa, 2021-09-07:
> Nice! Is this with the latest RPi kernel installed?

Yes

jerryneedell, 2021-09-07:
Cool! Worked for me as well!

makermelissa, 2021-09-08:
Ok, so getting the backlight to turn on by editing the DTS file hasn't been as easy as I thought.

Regarding the pinning, I was looking into that and the version than needs to be written to the pinning file is 1:1.20210527-1, but the file name is still uses 1.20210527-1. So I'm trying to see if there's an easy way to fix this without breaking compatibility.

jerryneedell, 2021-09-08:
Thanks for clarifying that.
If the backlight is fixed, then pinning is not necessary now, correct. It seems to be working ok as long as I manually turn on the backlight once after booting.

makermelissa, 2021-09-08:
Backlight is currently not fixed. The python script was just a test to see what the actual problem was. I'm just fixing pinning for the future and so people have an easy fix for now. I'm almost done updating the guide now and just need to update one more screenshot.

PR #192 (2021-09-09): Change backlight on ST7789 displays to be on by default
Hard to believe the fix was so simple, but when you run out of documented suggestions that don't seem to work and just start trying your own wacky ideas, sometimes it just works out that way.
Fixes #187. Fixes #189.
closed 2021-09-09 by makermelissa