adafruit/Raspberry-Pi-Installer-Scripts #188: AttributeError: type object 'gpiod.Chip' has no attribute 'OPEN_BY_NAME'

lucasgaley, 2021-09-02:
Following installation for Google Coral from:
https://learn.adafruit.com/circuitpython-on-google-coral-linux-blinka/initial-setup

I have followed this installation before, with no issues.

I have successfully followed all the steps with issues beginning when running "blinkatest.py". The traceback (full below) will happen if "board" or "digitalio" are imported. From a different forum, I saw there there may have been an update so I tried running the libgpiod.sh from a previous commit, but it didn't seem to make a difference (this part I'm not sure I did correctly).

Have attempted multiple times, any help would be appreciated!

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 150, in <module>
    from adafruit_blinka.board.coral_dev_board import *
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/board/coral_dev_board.py", line 3, in <module>
    from adafruit_blinka.microcontroller.nxp_imx8m import pin
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/nxp_imx8m/pin.py", line 4, in <module>
    I2C2_SCL = Pin(144)  # GPIO5_IO16
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 34, in __init__
    self._chip = gpiod.Chip("gpiochip0", gpiod.Chip.OPEN_BY_NAME)
AttributeError: type object 'gpiod.Chip' has no attribute 'OPEN_BY_NAME'

makermelissa, 2021-09-02:
Hi, can you see if there's a pre-compiled package available? See https://learn.adafruit.com/adding-a-single-board-computer-to-blinka/software-setup#installing-the-libgpiod-as-a-package-3058813-3 for how to check. If so, we should update the Coral guide to recommend that.

makermelissa, 2021-09-09:
Due to lack of response, I'm assuming this fixed worked and closing the issue. We can always reopen.

closed 2021-09-09 by makermelissa

lucasgaley, 2021-10-04:
Hi! Sorry for the delay, my project gets worked on in cycles. I had tried that solution previously, and all is working now. The mistake was on my end. I am using virtualenv, and I didn't properly include the site packages/install gpiod before using virtualenv. Thank you for your help!

lucasgaley, 2022-01-28:
It's been some time now, and I've had to reinstall on another separate Google Coral. I actually ran into this same issue again from a completely clean flash. Following the instructions again and ending up with the same error. After the error, I tried to install the pre-compiled package with no luck. Did a lot of installing and reinstalling, but something must have been off. Tried on three fresh flashes. From the documentation of the libgpiod that the installer script uses, the "OPEN_BY_NAME" attribute may have been removed a year ago now. 

I finally got it to work on a fresh flash with the pre-compiled libgpiod packages. I still followed the Adafruit initial setup instructions, but I replaced the entire "Install Libgpiod" section with the "Installing the libgpiod as a package" section from the link provided by @makermelissa.

Just in case this helps someone in the future. I do believe the instructional should be updated because following them to the letter multiple times did not work. And even if I uninstalled the libgpiod and used the pre-compiled, something was off and wouldn't work (possibly the bindings or some remnants). Hope this helps!