adafruit/Raspberry-Pi-Installer-Scripts #189: MiniPiTFT (240x135) no display (console install type)

nitz, 2021-09-05:
Hi all,

Recently decided to toss my MiniPiTFT on the cyberdeck bonnet. 

Using a fresh image of Raspberry Pi OS (Started from the regular 'Desktop' image via the official imager. Using the 32-bit flavor as this will eventually be used for a pi zero.)

Did an `apt full-upgrade`, `pip3 install`ed `adafruit-python-shell` & `click`. Cloned this repo, and ran `sudo python3 adafruit-pitft.py --display=st7789_240x135 --rotation=270 --install-type=console`

Install seemed to go fine. After reboot, there is still no display on the output. Using `adafruit-pitft.py` to uninstall and reinstall seems to make no difference either.

I'm not sure if this is because I'm trying to use it through the [cyberdeck bonnet](https://www.adafruit.com/product/4862) or if I'm just missing something else.

Thanks for taking a look!

```shell
pi@z:~ $ uname -a
Linux z 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux
pi@z:~ $ dmesg | grep st7789
[    6.811196] fb_st7789v: loading out-of-tree module taints kernel.
[    6.812108] fb_st7789v spi0.0: fbtft_property_value: width = 135
[    6.812136] fb_st7789v spi0.0: fbtft_property_value: height = 240
[    6.812172] fb_st7789v spi0.0: fbtft_property_value: buswidth = 8
[    6.812199] fb_st7789v spi0.0: fbtft_property_value: debug = 0
[    6.812232] fb_st7789v spi0.0: fbtft_property_value: rotate = 270
[    6.812259] fb_st7789v spi0.0: fbtft_property_value: fps = 60
[    6.989834] graphics fb0: fb_st7789v frame buffer, 240x135, 63 KiB video memory, 4 KiB buffer memory, fps=100, spi0.0 at 32 MHz
```

nitz, 2021-09-05:
In attempts to track down what's going on, I moved a PiTFT (240x240) & the SD card I had it running with to the Raspberry Pi 400, again through the cyberdeck bonnet, and it all worked exactly as expected. I moved the 240x135 & it's SD card to a Pi Zero that the 240x240 was running on, and experienced the same I saw on the Pi 400: The LCD backlight is lit, but there's no display.

I connected an HDMI display to the Pi Zero to see what output was visible there. It makes it through the entire boot process and the message about the auto-login displays, but (as expected, I supposed) input past that doesn't affect the HDMI display. I'm assuming it thinks it should be writing to the MiniPiTFT at that point. In fact, if I blindly type "sudo reboot", it does in fact reboot, as I'm able to see the shutdown/restart process on the HDMI display.

So it seems I either don't have the driver correctly communicating with the display, or I've got a dead screen, and I'm not quite sure how to diagnose either :)

makermelissa, 2021-09-06:
This sounds like it may be related to #175.

nitz, 2021-09-06:
I gave a shot adding changes to `minipitft114-overlay.dts` that looked like what was changed in 8bc1771bb2b1e40202058f39cc136620ffecfb39, (using GPIO 26 instead of 12 that was used by `st7789v_240x320-overlay.dts`), compiled and installed it, but didn't seem to make a difference. (I didn't dig into the commits around then to see if there were other changes that would need to be made too, though.)

closed 2021-09-09 by makermelissa

nitz, 2021-09-09:
Lmao, this is fantastic. Go figure 🤷‍♀️

PR #192 (2021-09-09): Change backlight on ST7789 displays to be on by default
Hard to believe the fix was so simple, but when you run out of documented suggestions that don't seem to work and just start trying your own wacky ideas, sometimes it just works out that way.
Fixes #187. Fixes #189.