adafruit/Raspberry-Pi-Installer-Scripts #190: /dev/spidev0.0 does not exist

FRobertAllen, 2021-09-08:
I follow the instructions on https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi

when I run
ls /dev/i2c* /dev/spi* 

I get this output
ls: cannot access '/dev/spi*': No such file or directory
 /dev/i2c-1

also when I run
python3 blinkatest.py

the output is
Hello blinka!
Digital IO ok!
I2C ok!
Traceback (most recent call last):
  File "blinkatest.py", line 16, in <module>
    spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
  File "/usr/local/lib/python3.7/dist-packages/busio.py", line 271, in __init__
    self._spi = _SPI(portId)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/spi.py", line 22, in __init__
    self._spi = spi.SPI(device=(portid, 0))
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PureIO/spi.py", line 167, in __init__
    raise IOError("{} does not exist".format(device))
OSError: /dev/spidev0.0 does not exist

makermelissa, 2021-09-08:
It means either SPI isn't enabled, which you can do manually with `raspi-config`, or you have a kernel driver installed that is using SPI. If you installed with PiTFT, be sure to run it and choose uninstall.

closed 2021-09-08 by makermelissa

FRobertAllen, 2021-09-08:
I dont see an uninstall prompt or option.
Is there a walkthrough other than through adafruit?