adafruit/Raspberry-Pi-Installer-Scripts #194: pitft 3.2 with raspberry pi zero 2 w

ro6er1, 2021-11-03:
I have a pitft 3.2 which seems to work fine with my Raspberry PI 4. However I just get a white screen when I try it with my PI Zero 2 w. I tried the easy install a few times and different ways: https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/easy-install-2 

But no joy. I was wondering if this is maybe because the PI Zero 2 w isn't support yet?

Roger

makermelissa, 2021-11-06:
We haven't tested it, but it should work. I noticed you linked to the 3.5" guide, but mentioned you have a 3.2" display. These 2 sizes use different drivers. Have you tried using the 2.8" resistive display selection? That should use the same one as the 3.2".

ro6er1, 2021-11-06:
Hi, sorry yes I tried the 2.8 drivers. I tried an image from the easy install page as well. And a few clean installs of raspberry os. But still no luck. Is there a debug script I can use or similar?

makermelissa, 2021-11-08:
It could be due to updates in Raspberry Pi OS. I'll have to test it out a bit later. I'm not aware of any debug script.

ro6er1, 2021-11-08:
Thanks, let me know if you find anything. It's really odd.

makermelissa, 2021-11-08:
I tested on a Pi Zero 2 W on latest release and am getting a black screen (same result on Pi 400), so there _is_ an issue with the latest release. If you're getting a white screen, I'd suggest double checking your solder connections. The white means it's getting powered, but not initialized. If you're still having issues, you can get additional help on https://forums.adafruit.com or discord. I'm going to open a new issue with regards to the black screen on the latest raspberry pi.

closed 2021-11-08 by makermelissa

makermelissa, 2021-11-09:
Once you fix your white screen, either use an older release of Raspberry Pi OS or the lite version.

ro6er1, 2021-11-09:
Thanks I had a second opinion on my solder job. It seems ok, I tested the gpio connection and I still have a white screen. I tried another pi zero 2 and still have a white screen. It's really odd.

Thanks for your help.

cajuncoder, 2023-01-28:
> I tested on a Pi Zero 2 W on latest release and am getting a black screen (same result on Pi 400), so there _is_ an issue with the latest release. If you're getting a white screen, I'd suggest [...]

Sorry, I see that you're having the same issue as I am with the Pi Zero W (first generation). I also get a black screen. Posted an issue here: https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/issues/239
Do none of the larger TFTs work with the Pi Zero anymore? What might be the issue?