adafruit/Raspberry-Pi-Installer-Scripts #195: 2.8" PiTFT black screen on Bullseye Raspberry Pi OS

makermelissa, 2021-11-08:
I haven't tested the other displays, but I'll start here. This was tested on the desktop version of Raspberry Pi OS using a Pi 400 and Raspberry Pi Zero 2 W with the FBCP method. Both used the same SD card, so I haven't tried other settings yet.

makermelissa, 2021-11-09:
Ok, success so far with the Pi 4 and Raspberry Pi OS Lite using FBCP. I'll try the desktop version again next. Maybe it has something to do with the new version using GTK+3 instead of GTK+2. This at least rules out the ILI9341 as the main culprit.

makermelissa, 2021-11-09:
Ok, I found this in the Bullseye notes (https://www.raspberrypi.com/news/raspberry-pi-os-debian-bullseye/):

> One of the changes which was prompted by the move to GTK+3 is that we are using a new window manager, called mutter, instead of the openbox window manager used in previous releases.
> 
> The window manager is responsible for drawing the title bar and window frame around each application window. Mutter is what is known as a compositing window manager, which means that rather than individually drawing the window frames straight onto the existing windows on screen, it draws all the windows to a memory buffer off screen, where it creates a complete image for the entire screen. This is then sent to the hardware to be displayed.
> 
> Because it has control of the whole screen, a compositing window manager can create some nice visual effects which are not possible (or at least not easy) with a traditional window manager: for instance, it can round the corners of windows, or display shaded borders behind each window. (While we did have rounded corners in openbox, this was done in a rather inefficient fashion by drawing multiple frames of varying sizes for each window – it worked, but it really wasn’t elegant!)

So it sounds like they have completely changed the way they draw, so FBCP is broken in this instance.

makermelissa, 2021-11-09:
Ok, I found a potential solution. Changing `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d` inside /boot/config.txt` seems to work for now. I got the idea from https://github.com/raspberrypi/firmware/issues/1647, so this may just be a temporary hiccup if they do fix it.

makermelissa, 2021-11-10:
It turns out just commenting it out altogether fixes touch.

closed 2021-11-10 by makermelissa

KalixtShawxo, 2022-01-13:
@makermelissa
Hi, I have been going round and round for a few days now trying to get my pttft 2.8 resistive screen working on a Raspi B+ running Bullseye based Raspberry PI OS from the imager v1.62, using FBCP and no matter what I have done, the HDMI output works, but the tft screen does not display anything in a mirror mode. Now, I can run the script for console mode, and it will display on the screen and the not HDMI...but now there is no GUI and no touchscreen.
I am trying to setup for use without a keyboard and mouse, so I kinda need the GUI.
I went so far as to even delete the existing adafruit-pitft.py file and create a new one with a copy/paste from your listed fix.
this was after following the instructions from Adafruit's "easy Install". Now I had at one point in the past gotten it to work with Buster as a mirror but the touchpoints were all off by 90* and I could never get it to align properly. And now with your fix I was hoping to try it again and use Bullseye. But I can't even get to the point of seeing if the touchscreen works at all, much less being misaligned like it was before.
Any help pointing me to a fix would be greatly appreciated.

makermelissa, 2022-01-14:
@KalixtShawxo please open a new issue for this. Thanks.

PR #197 (2021-11-10): PiTFT fix for bullseye
Fixes #195. Took a bit of trial and error, but this appears to be the fix for the PiTFT not working on Bullseye.