adafruit/Raspberry-Pi-Installer-Scripts #196: miniPiTFT 1.14" 240x135 not working after update

jrw429, 2021-11-09:
I have a Raspberry Pi 3B+ with a miniPiTFT 1.14" 240x135 display. I had previously set it up with the kernel modules and it worked fine. System output was displayed and interactive sessions worked fine. Then I did a full update of Raspbian. Rebooted and the display didn't work. Garbled micro-print on the display and not rotated. As expected - new kernel, need to re-run the installer. I re-ran the installer, rebooted, still same garbled micro-print. Then I updated my clone of the git repository for the Raspberry-Pi-Installer-Scripts. Re-ran the installer, rebooted, still garbled. Did some research, found the "force back to old kernel version" script, installed that, ran, re-ran installer, rebooted, still garbled.

At this point I don't see any other promising ideas. The display worked an hour ago, and now it is just garbled micro-print. Note that if I press enter on the USB keyboard, I can see new lines. So in some sense it is "working", but not intelligible
[image]
.

makermelissa, 2021-11-09:
Are you running lite or desktop version? If desktop, it's probably related to #195.

makermelissa, 2021-11-09:
It might be manifesting itself differently with the different display. Raspberry Pi often breaks things on both major and minor releases and doesn't give much notice beforehand. We suggest either using an older release for now or the lite version until we've had a chance to update the drivers. Unfortunately, there's no way to downgrade that I'm aware of.

jrw429, 2021-11-09:
Did a fresh install of Raspbian Lite on a different SD card. Installed all updates. Installed python3-pip. Then followed instructions on product page. Display is working again. Not sure of root cause of previous problem. I may try a fresh install of Raspbian Desktop and see if that still fails.

makermelissa, 2021-11-09:
Ok, glad you got it working. If you want desktop, you can check the other issue, where I think I found a potential solution. I'm going to close this for now in favor of the other one since that one is not display specific.

closed 2021-11-09 by makermelissa