adafruit/Raspberry-Pi-Installer-Scripts #203: python-spidev depends on python(<3.8) but 3.9 is to be installed

OfficialMuffin, 2021-12-13:


OfficialMuffin, 2021-12-13:
This error shows when running the adafruit-pitft.py file

makermelissa, 2022-03-07:
Please provide more information such as your specific setup in order to reproduce the issue. I just ran this on a Pi 4 with the latest version of Raspberry Pi and am not seeing the error:
```
pi@raspberrypi:~/Raspberry-Pi-Installer-Scripts $ sudo python3 adafruit-pitft.py --display=28r --rotation=90 --install-type=fbcp
This script downloads and installs
PiTFT Support using userspace touch
controls and a DTO for display drawing.
one of several configuration files.
Run time of up to 5 minutes. Reboot required!

Display Type: PiTFT 2.4", 2.8" or 3.2" resistive (240x320)
Rotation: 90
PITFT Checking init system...
Found systemd
/boot is mounted
PITFT System update
Updating apt indexes............
Reading package lists...
.........
PITFT Installing Python libraries & Software...
Installing Pre-requisite Software...This may take a few minutes!
PITFT Reading package lists...
PITFT Building dependency tree...
PITFT Reading state information...
PITFT bc is already the newest version (1.07.1-2+b1).
PITFT bc set to manually installed.
PITFT build-essential is already the newest version (12.9).
PITFT device-tree-compiler is already the newest version (1.6.0-1).
PITFT device-tree-compiler set to manually installed.
PITFT git is already the newest version (1:2.30.2-1).
PITFT python3-dev is already the newest version (3.9.2-3).
PITFT python3-dev set to manually installed.
PITFT python3-smbus is already the newest version (4.2-1+b1).
PITFT libraspberrypi-dev is already the newest version (1:2+git20211125~155417+14b90ff-3).
PITFT python3-pip is already the newest version (20.3.4-4+rpt1).
PITFT python3-spidev is already the newest version (20200602~200721-1).
PITFT The following package was automatically installed and is no longer required:
PITFT libfuse2
PITFT Use 'sudo apt autoremove' to remove it.
PITFT The following additional packages will be installed:
PITFT evemu-tools libevemu3
PITFT Suggested packages:
PITFT imagemagick
PITFT The following NEW packages will be installed:
PITFT evemu-tools evtest fbi libevemu3 libts-bin
PITFT 0 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.
PITFT Need to get 130 kB of archives.
PITFT After this operation, 468 kB of additional disk space will be used.
PITFT Get:1 http://mirror.pit.teraswitch.com/raspbian/raspbian bullseye/main armhf libevemu3 armhf 2.7.0-3 [11.5 kB]
PITFT Get:2 http://mirror.umd.edu/raspbian/raspbian bullseye/main armhf evemu-tools armhf 2.7.0-3 [14.3 kB]
PITFT Get:3 http://raspbian.mirror.constant.com/raspbian bullseye/main armhf evtest armhf 1:1.34-1 [13.7 kB]
PITFT Get:4 http://mirror.pit.teraswitch.com/raspbian/raspbian bullseye/main armhf fbi armhf 2.10-4 [53.0 kB]
PITFT Get:5 http://mirror.us.leaseweb.net/raspbian/raspbian bullseye/main armhf libts-bin armhf 1.22-1 [38.0 kB]
PITFT Fetched 130 kB in 3s (45.6 kB/s)
PITFT Selecting previously unselected package libevemu3:armhf.
(Reading database ... 100402 files and directories currently installed.)
PITFT Preparing to unpack .../libevemu3_2.7.0-3_armhf.deb ...
PITFT Unpacking libevemu3:armhf (2.7.0-3) ...
PITFT Selecting previously unselected package evemu-tools.
PITFT Preparing to unpack .../evemu-tools_2.7.0-3_armhf.deb ...
PITFT Unpacking evemu-tools (2.7.0-3) ...
PITFT Selecting previously unselected package evtest.
PITFT Preparing to unpack .../evtest_1%3a1.34-1_armhf.deb ...
PITFT Unpacking evtest (1:1.34-1) ...
PITFT Selecting previously unselected package fbi.
PITFT Preparing to unpack .../archives/fbi_2.10-4_armhf.deb ...
PITFT Unpacking fbi (2.10-4) ...
PITFT Selecting previously unselected package libts-bin.
PITFT Preparing to unpack .../libts-bin_1.22-1_armhf.deb ...
PITFT Unpacking libts-bin (1.22-1) ...
PITFT Setting up libevemu3:armhf (2.7.0-3) ...
PITFT Setting up libts-bin (1.22-1) ...
PITFT Setting up fbi (2.10-4) ...
PITFT Setting up evtest (1:1.34-1) ...
PITFT Setting up evemu-tools (2.7.0-3) ...
PITFT Processing triggers for gnome-menus (3.36.0-1) ...
PITFT Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u2) ...
PITFT Processing triggers for man-db (2.9.4-2) ...
PITFT Processing triggers for mailcap (3.69) ...
PITFT Processing triggers for desktop-file-utils (0.26-1) ...
PITFT Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
PITFT Collecting evdev
PITFT Downloading https://www.piwheels.org/simple/evdev/evdev-1.4.0-cp39-cp39-linux_armv7l.whl (83 kB)
PITFT Installing collected packages: evdev
PITFT Successfully installed evdev-1.4.0
PITFT Updating /boot/config.txt...
PITFT Updating SysFS rules for Touchscreen...
PITFT Updating TSLib default calibration...
PITFT Making sure console doesn't use PiTFT
Removing console fbcon map from /boot/cmdline.txt
Screen blanking time reset to 10 minutes
PITFT Adding FBCP support...
Installing cmake...
Downloading rpi-fbcp...
Uncompressing rpi-fbcp...
Building rpi-fbcp...
Installing rpi-fbcp...
Remove fbcp from /etc/rc.local, if it's there...
We have systemd, so install fbcp systemd unit...
PITFT Created symlink /etc/systemd/system/multi-user.target.wants/fbcp.service → /etc/systemd/system/fbcp.service.

Setting raspi-config to boot to desktop w/o login...
Configuring boot/config.txt for forced HDMI
PITFT Updating X11 default calibration...
PITFT Success!

Settings take effect on next boot.

REBOOT NOW? [Y/n] y
Reboot started...
```

makermelissa, 2022-06-01:
Closing due to lack of response and inability to reproduce the error.

closed 2022-06-01 by makermelissa