adafruit/Raspberry-Pi-Installer-Scripts #215: pttft 2.8 resistive screen not working on a Raspi B+

KalixtShawxo, 2022-03-05:
Sorry for the delay on opening this.....I had been sick for the past month or so. 
This is a followup/ repeat(?) issue from "PiTFT black screen on Bullseye Raspberry Pi OS Desktop (Issue #195)"

Hi, I have been going round and round for a few days now trying to get my pttft 2.8 resistive screen working on a Raspi B+ running Bullseye based Raspberry PI OS from the imager v1.62, using FBCP and no matter what I have done, the HDMI output works, but the tft screen does not display anything in a mirror mode. Now, I can run the script for console mode, and it will display on the screen and the not HDMI...but now there is no GUI and no touchscreen.
I am trying to setup for use without a keyboard and mouse, so I kinda need the GUI.
I went so far as to even delete the existing adafruit-pitft.py file and create a new one with a copy/paste from your listed fix.
this was after following the instructions from Adafruit's "easy Install". Now I had at one point in the past gotten it to work with Buster as a mirror but the touchpoints were all off by 90* and I could never get it to align properly. And now with your fix I was hoping to try it again and use Bullseye. But I can't even get to the point of seeing if the touchscreen works at all, much less being misaligned like it was before.
Any help pointing me to a fix would be greatly appreciated.

makermelissa, 2022-03-07:
I'll give it another look. Maybe they changed something since the last fix.

KalixtShawxo, 2022-03-08:
> The script is working fine on the latest version of buster.
>
> Things to keep in mind:
>
>    - If you must plug in HDMI and you are using a board with multiple HDMI outputs, make sure you are using HDMI0 (the one closest to the power plug).
>    - It should be installed with Standard Bullseye desktop using the Raspberry Pi Imager
>    - Follow this page: https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/easy-install-2
>    - Install for 2.8" Console with 90 degree rotation.
>    - Touch is working as expected
>    - If you need further help with your specific setup, please consider using the forums or discord.

Thanks Melissa, but I now give up on this. I ultimately need a console mode with touchscreen and not mirror mode, I just listed that I couldn't get either one to work correctly. Mirror mode would not show on the tft screen but did show on the monitor, while console mode showed on the tft but would not get to the gui not would the touchscreen work correctly. 
I can follow the directions perfectly and it does not work, must be hardware. Thanks again for your time, sorry to have wasted it. 
Roger.

makermelissa, 2022-03-08:
If you want touchscreen and console, you may want to look at Raspberry Pi Lite and use Mirror Mode. If it is in fact the hardware that is the issue, you should definitely try posting your issue on the forums.

makermelissa, 2022-03-12:
Hey, I think I might have been able to finally reproduce this issue with a Raspberry Pi 3 while working on another project. Which specific Model of Raspberry Pi are you using?

KalixtShawxo, 2022-03-14:
Ok, hopefully that means I'm not crazy here. 😊👍I've been trying with a 2014 model  B+ v 1.2

makermelissa, 2022-03-23:
Interestingly enough, the problem doesn't happen on the Pi Zero 2 W, but does on the Pi 3. So it's not related to 1 HDMI vs 2 HDMI ports like I initially thought.

makermelissa, 2022-03-23:
On a whim, I tried a different HDMI cable on the Pi 3 and it seems to be working now.

StephenGrable1, 2023-06-24:
I am also seeing this issue with my Raspberry pi B+ 2014 - explained more in [this post ](https://forums.adafruit.com/viewtopic.php?p=976892&hilit=2.8+touch#p976892) but it doesnt seem hardware related since the screen works fine displaying the "console" - its only when the install type is set to "fcbp" when the screen goes black after booting up