adafruit/Raspberry-Pi-Installer-Scripts #219: Error during installation

huba80, 2022-04-03:
This installation did not work for me. I get the following issues:

`Installing...
Reading package lists...
Building dependency tree...
Reading state information...
git is already the newest version (1:2.30.2-1).
raspberrypi-kernel-headers is already the newest version (1:1.20220328-1).
The following package was automatically installed and is no longer required:
libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 84 not upgraded.
fatal: destination path 'Raspberry-Pi-Installer-Scripts' already exists and is not an empty directory.

make[1]: *** /lib/modules/5.15.32-v8+/build: No such file or directory.  Stop.
make: *** [Makefile:38: clean] Error 2

make -C /lib/modules/5.15.32-v8+/build M=/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module clean
make[1]: *** /lib/modules/5.15.32-v8+/build: No such file or directory.  Stop.
make: *** [Makefile:35: all] Error 2

make -C /lib/modules/5.15.32-v8+/build M=/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module modules
make: *** No rule to make target 'snd-i2smic-rpi.ko', needed by 'install'.  Stop.

DONE.``

When I try to get a list by "arecord -l" then it returns an empty list. No device is shown.

Im on RPi 3B.

Thank you.

makermelissa, 2022-04-04:
Hi, which script is this for?

huba80, 2022-04-05:
Hi,

Im following this guide:
https://learn.adafruit.com/adafruit-i2s-mems-microphone-breakout/raspberry-pi-wiring-test

I reinstalled my Raspberry OS and tried again the same tutorial. I have managed to get the MEMS device in the device list.
Then I tried the next step:

`arecord -D plughw:1 -c1 -r 48000 -f S32_LE -t wav -V mono -v file.wav`

(My device number was 1)

Unfortunately during the test I get 00% on VU meter. The mic senses nothing. Any ide what could be wrong?

Thank you.

huba80, 2022-04-05:
I just found out that some of my GPIO pins (Mic DOUT to BCM 20 (pin 38) and Mic LRCL to BCM 19 (pin 35) are damaged.
Is there a way to change the GPIO pins within the script? If so may I ask you what file I need to edit and their location/path?
Thank you

makermelissa, 2022-04-05:
Oof. The issue is that the I2S hardware pins are hardware-based (see PCM-labeled pins at https://pinout.xyz/).I'm not aware of any way to remap them and have it done through software.

makermelissa, 2022-04-05:
Closing since the problem is not software related.

closed 2022-04-05 by makermelissa

Path-A, 2022-05-25:
For the record, I was having this same issue a few months ago with the Raspberry Pi 4 on Buster. The script would always complete properly after a reboot on the 2nd attempt. The issue seems to have gone away this month with the most recent Buster/Headers.