adafruit/Raspberry-Pi-Installer-Scripts #224: MiniPiTFT 1.14” not working on Ubuntu 22

Anamethatidk, 2022-06-05:
I am trying to use a adafruit MiniPiTFT with Raspberry Pi 4B on Ubuntu version 22.04 LTS. I am following the adafruit easy install instructions. Everything works until I put in “sudo python3 adafruit-pitft.py --display=st7789_240x135 --rotation=270 --install-type=fbcp”. The program then says “PiTFT Exiting due to error: /home/pi must be an existing directory (use -u /home/foo to specify)”. I have tried making a directory named pi and also one named home with another directory named pi inside of it but I still get the same error message. I am a beginner to Rasberry Pi and do not know how it works so if I am doing something wrong, which I probably am, I would really appreciate it if someone could help me figure this out.

Anamethatidk, 2022-06-06:
Thank you, I’ll try it.

Anamethatidk, 2022-06-06:
It works for the previous problem but then says “PITFT FATAL ERROR: Couldn’t open output file /boot/overlays/drm-minipitft114.dtbo: No such file or directory”. After a few more lines of updating, reading, and calculating, it says “PITFT Apt failed to install software!” Sorry if I am being annoying, I really appreciate the help.

makermelissa, 2022-06-06:
You can pass `--boot` along with the boot folder. According to #166 the folder on ubuntu is /boot/firmware, so you could add `--boot /boot/firmware` on now.

makermelissa, 2022-06-08:
Closing since all the needed items are in place.

closed 2022-06-08 by makermelissa