adafruit/Raspberry-Pi-Installer-Scripts #225: Does utilizing the Kernel Module install of the 1.3" mini PiTFT disable i2c access?

NickEngmann, 2022-06-09:
Currently utilizing the Kernel Module install of the 1.3" mini PiTFT display, and everything is working really well as far as display mannerisms go.

However, I am suffering to detect any of the i2c devices that I have plugged into the [Qwiic/STEMMA QT](https://www.adafruit.com/?q=stemma%20qt) connector on the mini PiTFT display.

I've ensured that i2c is turned on via `raspi-config`
I've successfully installed `adafruit-blinka` and have the examples working.
and `i2cdetect -y 1` returns nulls as shown below:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --    
```

`ls /dev/i2c* /dev/spi*` returns:

```
/dev/i2c-0   /dev/i2c-1   /dev/i2c-10   /dev/i2c-11
```

Board: `Raspberry Pi 2 Zero W`
kernel: `Linux raspberrypi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux`
OS: `Raspbian GNU/Linux 11 (bullseye)`

Relooking at both the install script and guides and can't find any documentation that confirms my theory that I can't utilize the kernel module and i2c at the same time, so thought it wouldn't hurt to ask.

makermelissa, 2022-06-09:
Hi, no it doesn't disable it and the fact you are seeing the `/dev/i2c` stuff confirms that. The STEMMA QT connector connects directly to the I2C pins of the Pi, so there's nothing fancy going on there. 

You try connecting your sensor directly to the I2C-1 pins of the Pi (see https://pinout.xyz/) just to test if it's maybe the connector on your Mini PiTFT or the stemma qt cable. If you're still not able to get it working, I'd recommend trying https://forums.adafruit.com/.

makermelissa, 2022-06-09:
Closing as this isn't related to the scripts.

closed 2022-06-09 by makermelissa