adafruit/Raspberry-Pi-Installer-Scripts #226: AttributeError: 'NoneType' object has no attribute 'glActiveTexture'

retikulumx, 2022-07-01:
On a Raspberry Zero, fresh installed, I get the following error

`Traceback (most recent call last):
  File "/boot/Pi_Eyes/eyes.py", line 9, in <module>
    import pi3d
  File "/usr/local/lib/python3.9/dist-packages/pi3d/__init__.py", line 10, in <module>
    from pi3d.constants import (USE_PYGAME, PIL_OK,
  File "/usr/local/lib/python3.9/dist-packages/pi3d/constants/__init__.py", line 198, in <module>
    PLATFORM, bcm, openegl, opengles = _detect_platform_and_load_libraries()
  File "/usr/local/lib/python3.9/dist-packages/pi3d/constants/__init__.py", line 193, in _detect_platform_and_load_libraries
    set_gles_function_args(opengles) #function defined in constants/gl.py
  File "/usr/local/lib/python3.9/dist-packages/pi3d/constants/gl.py", line 387, in set_gles_function_args
    gles.glActiveTexture.argtypes = [GLenum] #
AttributeError: 'NoneType' object has no attribute 'glActiveTexture'`

MNTrailBoss, 2022-10-16:
Ditto.    Right after reboot from the pieyes load.

PaintYourDragon, 2022-10-26:
From the forums, somewhat different hardware but same error, unsure of fix but something to try:

> Not sure what to suggest here. Just tested with 2022-09-22-raspios-buster-armhf-lite.img.xz, ran pi-eyes.sh for 240x240 screens, reboot when prompted and it runs successfully on both Pi 3 B and B+.
> 
> Just rampant speculation because I don’t know what else it could be, but if it’s a slower SD card, maybe some prerequisite service is taking some time to load. Try adding a delay (e.g. 10 seconds) in /etc/rc.local before eyes.py is run:
> `cd /boot/Pi_Eyes;sleep 10;python3 eyes.py --radius 240 &`

bikerglen, 2023-10-23:
FWIW: With a RPi 3B+, the Adafruit Animated Eyes Bonnet (# 3813), and two of the Adafruit OLED displays (# 1431), I had to go all the way back to `2021-05-07-raspios-buster-armhf-lite.img` to get it to work. With a more recent OS, e.g. any `Bullseye` or `Bookworm` version, I had the same `AttributeError: 'NoneType' object has no attribute 'glActiveTexture'` error as the other users.

No problem, however, using the latest and greatest OS images with a RPi 4.

retikulumx, 2023-10-26:
> No problem, however, using the latest and greatest OS images with a RPi 4.

You mean, on Rpi4 it runs with the most recent bookworm?

Yeah, my Zero with Pi-Eyes is also running on buster. It's a shame that the thing only runs with a Linux pre-pre-release. No further development. That has to be stated on the product pages and I would have liked to know this before buying the hardware.

bikerglen, 2023-10-26:
> You mean, on Rpi4 it runs with the most recent bookworm?

Crud. Sorry. I mean the latest legacy 32-bit full version. "Raspberry Pi OS (Legacy) Full" and dated 2023-05-03 in the official Raspberry Pi Imager tool. I can grab another OLED and double check this later today.

The Adafruit Pi Eyes installer script has trouble with the mandatory Python venv's in Bookworm.

PaintYourDragon, 2023-10-26:
Software dependencies are explained in the guide:
https://learn.adafruit.com/animated-snake-eyes-bonnet-for-raspberry-pi/software-installation
In short: **Desktop Bullseye** on Pi 4, **Lite Legacy** (Buster) on older boards, **32-bit** in all cases.
Haven’t yet tested w/Bookworm on any board (incl Pi 5).

Compatibility issues — the glActiveTexture message, etc. — are a product of the **Pi3D** library. That got all complicated with the release of the Pi 4 and is the reason different OS versions are needed in each case. A single-current-OS approach would be _really_ nice, but I don’t know if that’s ever coming to Pi3D or if broadly compatible alternatives exist (maybe vulkan, IDK).