adafruit/Raspberry-Pi-Installer-Scripts #227: Does not work for OSMC

blixten85, 2022-08-02:


ladyada, 2022-08-02:
its not guaranteed to work on OSMC but you'll get more help if you include details, error messages, screenshots, etc

blixten85, 2022-08-02:
[image]

makermelissa, 2022-08-02:
What are you running this on? Here's where it displays the error message: https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/i2samp.sh#L268-L272.

makermelissa, 2022-08-02:
Just after I asked, I realized the issue it's not working is because I don't think it's been tested on a 64-bit OS. It probably just needs some updating now that Raspberry Pi OS has an official 64-bit version.

blixten85, 2022-08-02:
Maybe.

It's OSMC on a raspberry pi 4B. Arm64

PR #231 (2022-10-31): Updated i2samp to run on 64-bit Raspberry Pi OS
Fixes #227.
closed 2022-10-31 by makermelissa