adafruit/Raspberry-Pi-Installer-Scripts #228: Adafruit Mini PiTFT 1.3" - 240x240 TFT, console image offset when rotated 90° or 270°

grenius, 2022-08-11:
When using the pi-tft installer and choosing either 90 or 270° the console image is offest and the "old" console remains on the screen in it's original orientation.

grenius, 2022-08-14:
Environment: raspberry pi OS buster

Hardware: raspberry pi zero 2 W / Zero2Go UPS hat / Zero4U USB hub

makermelissa, 2022-08-15:
Thanks for the details. The only one of those that I have is the Pi Zero 2 W.

makermelissa, 2022-11-03:
Verified with a Pi Zero 2 W using the Desktop installation of Buster, Console install option, and a BrainCraft HAT. I suspect the board doesn't matter and I also suspect it may do the same on bullseye, but perhaps not since it has newer drivers. First one is easy to verify.

grenius, 2022-11-03:
Thanks Melissa, I just borked my zero with a power failure while updating
so I'll reinstall with bullseye.

makermelissa, 2022-11-03:
Verified it still happens with bullseye and even on lite install. If you can, I'd recommend lite and using FBCP and it should be fine for now.

makermelissa, 2022-11-03:
Hmm, I thought I had previously fixed this. I found #129, which changes the driver parameter from `rotate` to `rotation`. This appears to fix it. I need to investigate why it was changed back.

makermelissa, 2022-11-03:
It looks like it was changed in #157 by me. Unfortunately I never documented why I made that change. My guess it was either:
1. It broke something for the FBCP install (possibly issue #138)
2. It was an accident.

I need to test to see if there are issues with the desktop if it uses `rotation` instead of `rotate`. We may just simply need to have alternate parameters for the console install if it is an issue.

makermelissa, 2022-11-03:
> Thanks Melissa, I just borked my zero with a power failure while updating so I'll reinstall with bullseye.

For now, you can edit `/boot/config.txt` and just manually change the parameter name.

PR #233 (2022-11-04): Fix rotation parameter for ST7789 displays
Fixes #228.

Apparently this was done in #129 and then reverted for some reason in #157. Tested with console and fbcp and it seems to work fine on both.

@ladyada
closed 2022-11-04 by makermelissa