adafruit/Raspberry-Pi-Installer-Scripts #241: rotation vs rotate

mikeysklar, 2023-02-17:
### Script Command

```shell
sudo script python3 adafruit-pitft.py --display=st7789_240x135 --rotation=270 --install-type=console
```

### Operating System

32-bit Buster Lite

### Hardware

Pi 4B

### Behavior

rotate works with the 1.14” Mini PiTFT in the /boot/config.txt, but rotation does not. 

### Description

A patch was added in November to the adafruit-pitft.py installer to use the word rotation. 

https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/pull/233/files/2f8e05b9fc61a2f4be25d3579de02c02617bec99

This was probably a change compatible with newer kernels that Bullseye uses, but is not compatible with Buster.

makermelissa, 2023-02-17:
I think this may be a case of updating the documentation as I believe the patch is correct.

mikeysklar, 2023-02-17:
A forum user had made their setup work after swapping out s/rotation/rotate in their config.txt. I've ordered a 1.14 PiTFT and will verify.

https://forums.adafruit.com/viewtopic.php?t=198749

makermelissa, 2023-02-17:
Thanks. I meant it's likely the guide itself probably needs to just be changed to have the user use `rotate` instead of `rotation`. I'm just not sure if that's for all boards or just this one.

mikeysklar, 2023-02-17:
I see. Yes, it might be more appropriate to just change the guide. I'll try some combinations on the displays I have to confirm behavior.

makermelissa, 2023-05-12:
I started to change the guide, but then realized that was the parameter for the script and not for config.txt. I ended up changing it in the install script.

PR #249 (2023-05-19): Change rotation to rotate for ST7789 displays
Fixes #241.

After some review, it looks like guide updates were not the right approach, but just changing how it writes the parameter into config.txt for the ST7789 displays only.
closed 2023-05-19 by makermelissa