adafruit/Raspberry-Pi-Installer-Scripts #242: Broken PiTFT 240x135 Installer

timptner, 2023-03-18:
### Script Command

```shell
sudo python3 adafruit-pitft.py --display=st7789_240x135 --rotation=270 --install-type=console
```

### Operating System

64-bit Raspberry Pi OS Lite with Bullseye

### Hardware

Raspberry Pi 4B

### Behavior

```
This script downloads and installs
PiTFT Support using userspace touch
controls and a DTO for display drawing.
one of several configuration files.
Run time of up to 5 minutes. Reboot required!

Display Type: MiniPiTFT 1.14" display
WARNING! WILL UPGRADE YOUR KERNEL TO LATEST
Rotation: 270
PITFT Checking init system...
Found systemd
/boot is mounted
PITFT System update
Updating apt indexes............
Reading package lists...
.........
PITFT Installing Python libraries & Software...
Installing Pre-requisite Software...This may take a few minutes!
PITFT Reading package lists...
PITFT 
PITFT Building dependency tree...
PITFT 
PITFT Reading state information...
PITFT 
PITFT bc is already the newest version (1.07.1-2+b2).
build-essential is already the newest version (12.9).
device-tree-compiler is already the newest version (1.6.0-1).
evtest is already the newest version (1:1.34-1).
fbi is already the newest version (2.10-4).
python3-smbus is already the newest version (4.2-1+b1).
python3-dev is already the newest version (3.9.2-3).
libts-bin is already the newest version (1.22-1+b1).
git is already the newest version (1:2.30.2-1+deb11u2).
libraspberrypi-dev is already the newest version (1:2+git20220616~133208+6e8f786-1).
python3-pip is already the newest version (20.3.4-4+rpt1+deb11u1).
python3-spidev is already the newest version (20200602~200721-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
PITFT Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
PITFT Requirement already satisfied: evdev in /usr/local/lib/python3.9/dist-packages (1.6.1)
PITFT Installing display drivers and device tree overlay...
Compiling Device Tree Overlay
############# UPGRADING KERNEL ###############
Updating packages...
Upgrading packages...
PITFT Reading package lists...
PITFT 
PITFT Building dependency tree...
PITFT Reading state information...
PITFT 
PITFT Calculating upgrade...
PITFT 
PITFT 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Installing Kernel Headers. This may take a few minutes...
Compiling and installing display driver...
PITFT make -C /lib/modules/`uname -r`/build M=/home/aiven/Raspberry-Pi-Installer-Scripts/st7789_module modules
PITFT make[1]: Entering directory '/usr/src/linux-headers-6.1.19-v8+'
PITFT CC [M]  /home/aiven/Raspberry-Pi-Installer-Scripts/st7789_module/fb_st7789v.o
PITFT In file included from /home/aiven/Raspberry-Pi-Installer-Scripts/st7789_module/fb_st7789v.c:15:
/home/aiven/Raspberry-Pi-Installer-Scripts/st7789_module/fbtft.h:326:12: error: initialization of ‘void (*)(struct spi_device *)’ from incompatible pointer type ‘int (*)(struct spi_device *)’ [-Werror=incompatible-pointer-types]
  326 |  .remove = fbtft_driver_remove_spi,                                 \
      |            ^~~~~~~~~~~~~~~~~~~~~~~
/home/aiven/Raspberry-Pi-Installer-Scripts/st7789_module/fbtft.h:361:2: note: in expansion of macro ‘FBTFT_REGISTER_DRIVER_END’
  361 |  FBTFT_REGISTER_DRIVER_END(_name, _display)
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/aiven/Raspberry-Pi-Installer-Scripts/st7789_module/fb_st7789v.c:314:1: note: in expansion of macro ‘FBTFT_REGISTER_DRIVER’
  314 | FBTFT_REGISTER_DRIVER(DRVNAME, "sitronix,st7789v", &display);
      | ^~~~~~~~~~~~~~~~~~~~~
PITFT /home/aiven/Raspberry-Pi-Installer-Scripts/st7789_module/fbtft.h:326:12: note: (near initialization for ‘fbtft_driver_spi_driver.remove’)
  326 |  .remove = fbtft_driver_remove_spi,                                 \
      |            ^~~~~~~~~~~~~~~~~~~~~~~
/home/aiven/Raspberry-Pi-Installer-Scripts/st7789_module/fbtft.h:361:2: note: in expansion of macro ‘FBTFT_REGISTER_DRIVER_END’
  361 |  FBTFT_REGISTER_DRIVER_END(_name, _display)
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/aiven/Raspberry-Pi-Installer-Scripts/st7789_module/fb_st7789v.c:314:1: note: in expansion of macro ‘FBTFT_REGISTER_DRIVER’
  314 | FBTFT_REGISTER_DRIVER(DRVNAME, "sitronix,st7789v", &display);
      | ^~~~~~~~~~~~~~~~~~~~~
PITFT cc1: some warnings being treated as errors
PITFT make[2]: *** [scripts/Makefile.build:250: /home/aiven/Raspberry-Pi-Installer-Scripts/st7789_module/fb_st7789v.o] Error 1
PITFT make[1]: *** [Makefile:2012: /home/aiven/Raspberry-Pi-Installer-Scripts/st7789_module] Error 2
PITFT make[1]: Leaving directory '/usr/src/linux-headers-6.1.19-v8+'
PITFT make: *** [Makefile:6: default] Error 2
PITFT Apt failed to compile ST7789V drivers!
```

Niwreg, 2023-03-20:
Samething happens on an Raspberry Pi Model B Rev 2 with 32bit Raspbian GNU/Linux 11 (bullseye)****

ubopod, 2023-03-28:
I had the same issue with ST7789 display on Adafruit Braincraft HAT. I tested it on both 64-bit and 32-bit Raspberry Pi OS Desktop with Bullseye

> uname -r
6.1.19-v8+

> dpkg -l raspberrypi-kernel-headers
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                       Version        Architecture Description
+++-==========================-==============-============-==============================================
ii  raspberrypi-kernel-headers 1:1.20230317-1 arm64        Header files for the Raspberry Pi Linux kernel

makermelissa, 2023-03-28:
Ok, it looks like Raspberry Pi did an update recently, which is causing the display drivers to not compile anymore. This sort of thing actually happens quite regularly.

ubopod, 2023-03-29:
It seems like that the fbdev graphics subsystem is becoming obsolete and also the drivers that depend on them. The newer drivers are DRM-based:

[List of LCDs with supported DRM-drivers](https://github.com/notro/fbtft/wiki/DRM-drivers)

* ST7789 controller which is used by Brain Craft HAT is not on the list yet. 

I guess one of the limitations of using framebuffer copy (fbcp) is that it shares the same buffer as the main display (copies from /dev/fb0) and it is not possible to preserve graphics function on both HDMI and LCD display (the full desktop is lost on Raspberry Pi for example). 

I guess with DRM approach, the hardware-accelerated video rendering pipeline can be bypassed and write directly on the SPI bus from memory and still prefer the Desktop display? 

Anyways, if you need any help with testing and debugging please do let me know. I am investigating solutions here but my knowledge of graphics subsystems is very basic...

ubopod, 2023-03-30:
Thanks for the pointer Melisa! This was super helpful. Running the following command did the trick and I can now see the desktop on my LCD on the latest 32-bit Rasbian OS (`uname -r 6.1.19-v8+`)

`sudo python3 adafruit-pitft-mipi.py --display=st7789_240x240 --rotation=0`

I am going to test this on 64-bit OS and post an update here. Also, I am not using touch (non-touch display on Brain Craft HAT) and can't tell if touch would work or not...
I am trying to preserve the HDMI desktop on my Raspberry Pi and develop a GUI using either Qt or Kivy and display it independently on the LCD. So far I haven't had any luck, but since the driver uses DRM buffer and does not rely on Video Core stuff this should be doable right? I would appreciate any pointers in this.

makermelissa, 2023-03-30:
Awesome. It should work on 64-bit. I like to have stuff working on both because I occasionally need either for projects. As for whether it should work, it's been a while since I took a close look so I really don't remember the details with certainty, but I believe that's correct.

I'll take a closer look at this in the next few days, but hopefully that can get you temporarily unstuck. When I mentioned touch, I was referring to the larger 2.8" displays, but the resistive touch chips are obsolete anyways (and unobtainable anymore, which is why those displays have not been in stock in months). The way I was getting around the touch issue was writing a separate DTO for that, but also found some deeper issues when I did that. Anyhow, we're just going to focus on the ST7789 displays (which are non-touch) with this...

makermelissa, 2023-04-04:
This is very weird. I installed the 32-bit Raspberry Pi OS via the images. Then did an update/upgrade. Then I tried installing the pitft script and was getting a "PITFT Kernel was updated, but needs to be loaded. Please reboot now and re-run script!" error along the way even after rebooting.

After digging in a bit, I ended up doing a `uname -a` and it comes back with `Linux raspberrypi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux`. The aarch64 means it was converted over to 64-bit at some point along the update path.

FYI, the reason I was getting the PiTFT error was because it was checking for a build folder under the release. The build folder exists, except under the 32-bit kernel and the script did not expect a 32-bit to 64-bit conversion.

makermelissa, 2023-04-04:
Ok, after more digging, I have discovered the upgrade happens after the apt update/upgrade. According to https://github.com/raspberrypi/firmware/issues/1795 this behavior is intentional. It can be manually reverted to 32-bit by adding `arm_64bit=0` to the config.txt file, but that's not really what we want. So from here on out, I'll just test this on 64-bit.

makermelissa, 2023-04-04:
Ok, I think this may be a bug on Raspberry Pi's end. Installing raspberrypi-kernel-headers on a aarch64 kernel should install aarch64 headers. Instead it's installing armhf (32-bit) headers.

makermelissa, 2023-04-04:
Yep, it is. It looks like somebody already filed a bug report at https://github.com/raspberrypi/linux/issues/5408.

makermelissa, 2023-04-04:
Ok, I found running `dpkg --print-architecture` shows **armhf**, which is the reason for the wrong headers. I'm still trying to decide on the best approach to fixing this. I supposed it depends on whether it compiles as armhf or aarch64.

makermelissa, 2023-04-04:
With some temporary workarounds(set `export ARCH=arm` and hardcoding the Makefile to use the headers in the 7l+ folder), I was able to reproduce the original issue by trying to compile directly. I think I recall running into this during the 4->5 kernel upgrade (fixed in #157) and we're on 6 now. So the other stuff is a completely different issue that will need to be addressed in another issue.

makermelissa, 2023-04-04:
Ok, I found a potential workaround. There's a built-in fbtft driver. I set up the device tree overlay and set it to use the fkms graphics and it showed to boot screen for a few seconds, but then stopped rendering anything.

makermelissa, 2023-04-04:
So this issue looks like it will be fairly easy to fix. It appears compiling and installing the driver is no longer necessary. Installing the device tree overlay just works now.

closed 2023-04-05 by makermelissa