adafruit/Raspberry-Pi-Installer-Scripts #243: License Clarification for i2s_mic_module files

jalapenopuzzle, 2023-03-21:
I request licensing clarification for the following files:

Raspberry-Pi-Installer-Scripts/i2smic.py
Raspberry-Pi-Installer-Scripts/i2s_mic_module/Makefile
Raspberry-Pi-Installer-Scripts/i2s_mic_module/README.md
Raspberry-Pi-Installer-Scripts/i2s_mic_module/dkms.conf
Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c
Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.h

Thank you very much for making these awesome resources available!

jalapenopuzzle, 2023-03-21:
Oh see that snd-i2smic-rpi.c contains the line
MODULE_LICENSE("GPL v2");

Unfortunately it is not obvious whether this also applies to all the other files listed.

makermelissa, 2023-03-21:
Good call. I think it's the same for everything in the i2s_mic_module folder. Forthe installer script itself, `i2smic.py` is based on https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/converted_shell_scripts/i2smic.sh, which didn't specify a license, so I'm just going to say the MIT license, which is what we pretty much put on all of the stuff we write.

Anyways, having licensing headers on everything (along with an automatic check) is a good idea. I created an issue for adding this.