adafruit/Raspberry-Pi-Installer-Scripts #247: i2smic.py script FAILS with latest Linux Kernel 6.1.21-v8+

alexcorvis84, 2023-04-11:
### Script Command

```shell
Running the *i2smic.py* fails to build and install the module if you're running the latest Linux Kernel 6.1.21-v8+

[Script](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/i2smic.py#L40) fails when tries to do the make command.
```

### Operating System

32-bit Raspberry Pi Os Lite with Bullseye

### Hardware

RaspberryPi 4B

### Behavior

Cloning into 'Raspberry-Pi-Installer-Scripts'...
make -C /lib/modules/6.1.21-v8+/build M=/home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module clean
make[1]: *** /lib/modules/6.1.21-v8+/build: No such file or directory.  Stop.
make: *** [Makefile:38: clean] Error 2
make -C /lib/modules/6.1.21-v8+/build M=/home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module modules
make[1]: *** /lib/modules/6.1.21-v8+/build: No such file or directory.  Stop.
make: *** [Makefile:35: all] Error 2
make: *** No rule to make target 'snd-i2smic-rpi.ko', needed by 'install'.  Stop.
DONE.

Settings take effect on next boot.

REBOOT NOW? [Y/n]

alexcorvis84, 2023-04-12:
Looking into the issue, I found that script didn´t work due it was not able to found the right directory **/lib/modules/6.1.21-v8+/build**

Not sure why, but in a fresh install, following the guide mentioned previously, after updating the raspberry with update/upgrade, you can can check that version bumped to **6.1.21-v8+** (from original RPi OS Lite 32-bit -> 5.15.84-v7l+).

I could check that once script installs [raspberrypi-kernel-headers](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/i2smic.py#L34), it installs headers in **lib/modules/**: 
```
6.1.21+/
6.1.21-v7+/
6.1.21-v7l+/
6.1.21-v8+/ 
```
But 6.1.21-v8+/ **does not** contains **/build** folder whereas the rest do.

I tried to modify the [Makefile MDIR & KDIR](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/i2s_mic_module/Makefile#L28) pointing to an existing headers files (e.g 6.1.21-v7l+/)
```
KDIR	?= /lib/modules/6.1.21-v7l+/build
MDIR	?= /lib/modules/6.1.21-v7l+/
```

But script failed with the following log:

```
This script downloads and installs
I2S microphone support.

RASPBERRY_PI_4B detected.

Auto load module at boot? [y/n] y 

Installing...
Reading package lists...

Building dependency tree...

Reading state information...

git is already the newest version (1:2.30.2-1+deb11u2).
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
fatal: destination path 'Raspberry-Pi-Installer-Scripts' already exists and is not an empty directory.
make -C /lib/modules/6.1.21-v7+/build M=/home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module clean
make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v7+'
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v7+'
make -C /lib/modules/6.1.21-v7+/build M=/home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v7+'
CC [M]  /home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.o
In file included from ./include/linux/build_bug.h:5,
                 from ./include/linux/container_of.h:5,
                 from ./include/linux/list.h:5,
                 from ./include/linux/module.h:12,
                 from /home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:18:
./include/linux/compiler.h:246:10: fatal error: asm/rwonce.h: No such file or directory
  246 | #include <asm/rwonce.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.o] Error 1
make[1]: *** [Makefile:2012: /home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v7+'
make: *** [Makefile:35: all] Error 2
make: *** No rule to make target 'snd-i2smic-rpi.ko', needed by 'install'.  Stop.
DONE.

Settings take effect on next boot.

REBOOT NOW? [Y/n] n
Exiting without reboot.
```
@makermelissa @caternuson could you please review and check?

The script seems to have conflicts with the latest released kernel versions.

I have tried looking for alternative methods to get the hardware working, but they all refer to the same sources and script published by Adafruit. 

I would therefore be grateful if it can be fixed as soon as possible :-)

Thanks in advance.

makermelissa, 2023-04-12:
This sounds similar to what I encountered in #242. What's happening is Raspberry Pi is upgrading the kernel to 64-bit, but leaving the user space at 32-bit. Then when it tries to install the kernel headers, it only installs the 32-bit ones and not the 64 bit. This was by design and a huge change that was different than anything Raspberry Pi had done before, so the guide is likely describing something that we weren't expecting.

With regards to the display driver, the answer ended up being to not worry about compiling it and just use a built-in one. I wonder if something similar can be done here. For the time being, you may want to try installing the 64-bit version of Raspberry Pi.

makermelissa, 2023-04-12:
There doesn't appear to be a simple solution like with the display driver. If you do find it works on 64-bit, I can modify the script to check if it is 32-bit user space and a 64-bit kernel and make some suggestions to the user:
- Reinstall 64-bit
- Setting it to 32-bit more by adding `arm_64bit=0` to `/boot/config.txt`

This would be more of a temporary fix until Raspberry Pi can figure out a better solution (see https://github.com/raspberrypi/firmware/issues/1795).

Additionally, it would be nice to see if it compiles in 32-bit. On the Pi 4B, it should be the `6.1.21-v7l+` kernel rather than `6.1.21-v7+`. I figured this out because I had done a `uname -a` on a fresh install prior to updating and it was using the v7l+ kernel.

alexcorvis84, 2023-04-13:
> This sounds similar to what I encountered in #242. What's happening is Raspberry Pi is upgrading the kernel to 64-bit, but leaving the user space at 32-bit. Then when it tries to install the kernel headers, it only installs the 32-bit ones and not the 64-bit. This was by design (on Raspberry Pi's part) and a huge change that was different than anything Raspberry Pi had done before, so the guide is likely describing something that we weren't expecting.
> 
> With regards to the display driver, the answer ended up being to not worry about compiling it and just use a built-in one. I wonder if something similar can be done here. For the time being, you may want to try installing the 64-bit version of Raspberry Pi.

Hi @makermelissa, 

First of all, thank you so much for your quick review and response :-) !

Thanks for sharing this further info that let me understand better what's happening...

Regarding installing the 64-bit version of RPi OS, I can't as I'm using a software that needs to be run on RPi OS 32-bit.

alexcorvis84, 2023-04-13:
> There doesn't appear to be a simple solution like with the display driver. If you do find it works on 64-bit, I can modify the script to check if it is 32-bit user space and a 64-bit kernel and make some suggestions to the user:
> 
>     * Reinstall 64-bit
> 
>     * Setting it to 32-bit more by adding `arm_64bit=0` to `/boot/config.txt`
> 
> 
> This would be more of a temporary fix until Raspberry Pi can figure out a better solution (see [raspberrypi/firmware#1795](https://github.com/raspberrypi/firmware/issues/1795)).
> 

Thanks for your feedback.

For the time being, I've been able to compile the module in a fresh RPi OS 32bit install following the below steps:

- I **did not** perform `sudo apt upgrade`, as I checked it updates the bootloader and kernel headers to latest versions.
```
raspberrypi-bootloader/stable 1:1.20230405-1 armhf [upgradable from: 1:1.20230106-1]
raspberrypi-kernel-headers/stable 1:1.20230405-1 armhf [upgradable from: 1:1.20230106-1]
```
- I found the package 📦 which contains the [headers for version 5.15.84-v7l+](http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20230106-1_armhf.deb)
- Installing this package with `sudo dpkg -i raspberrypi-kernel-headers_1.20230106-1_armhf.deb`, which matches the RPi OS Lite 32-bit kernel version, you get the headers needed to compile and install the module matching the kernel version from latest RPi OS 32bit released.
- In order to let do the script installation without errors, I modified the original script, removing the [raspberrypi-kernel-headers package](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/i2smic.py#L34) from the command.
- I can confirm this **workaround** worked and module was compiled successfully without any errors.
```

Cloning into 'Raspberry-Pi-Installer-Scripts'...
make -C /lib/modules/5.15.84-v7l+/build M=/home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module clean
make[1]: Entering directory '/usr/src/linux-headers-5.15.84-v7l+'
make[1]: Leaving directory '/usr/src/linux-headers-5.15.84-v7l+'
make -C /lib/modules/5.15.84-v7l+/build M=/home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.84-v7l+'
CC [M]  /home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.o
MODPOST /home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module/Module.symvers
CC [M]  /home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.mod.o
LD [M]  /home/ansible/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.15.84-v7l+'
rm -f /lib/modules/5.15.84-v7l+/kernel/sound/drivers/snd-i2smic-rpi.ko
install -m644 -b -D snd-i2smic-rpi.ko /lib/modules/5.15.84-v7l+/kernel/sound/drivers/snd-i2smic-rpi.ko
depmod -a
DONE.

Settings take effect on next boot.

REBOOT NOW? [Y/n] Y
Reboot started...
```

> Additionally, it would be nice to see if it compiles in 32-bit. On the Pi 4B, it should be the `6.1.21-v7l+` kernel rather than `6.1.21-v7+`. I figured this out because I had done a `uname -a` on a fresh install prior to updating and it was using the v7l+ kernel.

I tried it as well yesterday (with `6.1.21-v7l+`) and it didn't compile due the **same error** shared on my [second post](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/issues/247#issuecomment-1505003555).

I'll let you know any feedback If I try to compile it with a RPi OS 64bit.

Meanwhile, I agree with you to apply the temporary fix you mentioned previously to let other users skip the error or my workaround might help you indeed.

alexcorvis84, 2023-04-13:
@makermelissa FYI

> I'll let you know any feedback If I try to compile it with a RPi OS 64bit.

I've just tried to install a fresh **RPi OS 64bit image** to test.

After issuing the sudo apt-get update/upgrade commands, `uname -a ` output:

`Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`

If you list the /lib/modules installed, there's **only** 6.1.21-v8+/ listed and this are the contents:

[image]

So, **script will fail** as the compilation [Makefile](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/i2s_mic_module/Makefile#L28) is missing the **/build** path files needed to compile.

Might be the package of _raspberrypi-kernel-headers_ has changed with the bump to 64-bit kernel and compilation should be done in a different way? 🤔 

Let me know if I can assist with any other tests.

makermelissa, 2023-04-13:
Thank you SO much for testing this out. It helps immensely.

alexcorvis84, 2023-04-14:
> If you manually download the version-matching **arm64** kernel header package from http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/ and manually install that, does it work? The pi people should switch kernel headers with kernels together.

If you're running a Raspberry Pi OS 64bit, YES it should work. [I tested it yesterday](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/issues/247#issuecomment-1507025083) on a fresh install, and after installing the headers, the script run without issues.

Please note I installed manually the raspberrypi-kernel-headers manually, but script [should do the work](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/i2smic.py#L34) for you as well.

Please let us know any feedback.

makermelissa, 2023-04-14:
> If you manually download the version-matching **arm64** kernel header package from http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/ and manually install that, does it work? The pi people should switch kernel headers with kernels together.

First of all, thanks for the links. I wasn't sure where those were kept.

> Please note I installed manually the raspberrypi-kernel-headers manually, but script [should do the work](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/i2smic.py#L34) for you as well.

I'm guessing @HinTak is referring to the case when the 32-bit OS is in place with the 64-bit kernel upgrade (without the `arm_64bit=0` flag set). The question is then is: under those conditions, can we manually install the 64-bit headers (with the above link) and compile successfully?

If you'd like to test, that would be awesome. If not, I'll get to it soon.

alexcorvis84, 2023-04-18:
> 

> > If you manually download the version-matching **arm64** kernel header package from http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/ and manually install that, does it work? The pi people should switch kernel headers with kernels together.
> 
> First of all, thanks for the links. I wasn't sure where those were kept.
> 
> > Please note I installed manually the raspberrypi-kernel-headers manually, but script [should do the work](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/i2smic.py#L34) for you as well.
> 
> I'm guessing @HinTak is referring to the case when the 32-bit OS is in place with the 64-bit kernel upgrade (without the `arm_64bit=0` flag set). The question is then is: under those conditions, can we manually install the 64-bit headers (with the above link) and compile successfully?
> 
> If you'd like to test, that would be awesome. If not, I'll get to it soon. I agree the script should do the work. We're mostly figuring out what we want the script to do differently at this point. ;)

Hi @makermelissa 

I've just tried and it does not work. I got the following error:

```
ansible@raspberrypi:~ $ sudo dpkg -i raspberrypi-kernel-headers_1.20230405-1_arm64.deb 
dpkg: error processing archive raspberrypi-kernel-headers_1.20230405-1_arm64.deb (--install):
 package architecture (arm64) does not match system (armhf)
Errors were encountered while processing:
 raspberrypi-kernel-headers_1.20230405-1_arm64.deb
```

Please review and check below log:

```
ansible@raspberrypi:~ $ uname -m
aarch64
ansible@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
ansible@raspberrypi:~ $ cat /boot/config.txt 
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

# Enable UART2 & UART4
dtoverlay=uart2
dtoverlay=uart4

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
enable_uart=1
ansible@raspberrypi:~ $ sudo dpkg -i raspberrypi-kernel-headers_1.20230405-1_arm64.deb 
dpkg: error processing archive raspberrypi-kernel-headers_1.20230405-1_arm64.deb (--install):
 package architecture (arm64) does not match system (armhf)
Errors were encountered while processing:
 raspberrypi-kernel-headers_1.20230405-1_arm64.deb
ansible@raspberrypi:~ $ 
```

Please let me know if I can help you with any other test.

makermelissa, 2023-04-18:
Thanks for checking. It appears the 32-bit user space with a 64-bit kernel is a bit of a no man's land in regards to compiling at the moment.

closed 2023-05-19 by makermelissa

KePere, 2024-06-17:
Hi to everyone!

I've read all your comments and it sounds me really tricky.
I am not an expert on Python and right now I am trying to install adafruit MEMS microphone, same as previous messages, to Bullseye 64-bit Full SO Raspberry Pi 4B.
The error is related with the following lines:
-------------
This script downloads and installs
I2S microphone support.

RASPBERRY_PI_4B detected.

Auto load module at boot? [y/n] y

Installing...
S'està llegint la llista de paquets...

S'està construint l'arbre de dependències...

S'està llegint la informació de l'estat...

git ja està en la versió més recent (1:2.30.2-1+deb11u2).
raspberrypi-kernel-headers ja està en la versió més recent (1:1.20230405-1).
El paquets següents s'han instal·lat automàticament i ja no són necessaris:
libfuse2 libwpe-1.0-1 libwpebackend-fdo-1.0-1
Empreu «sudo apt autoremove» per a suprimir-los.
0 actualitzats, 0 nous a instal·lar, 0 a suprimir i 0 no actualitzats.
fatal: el BANNED BANNED «Raspberry-Pi-Installer-Scripts» ja existeix i no és un directori buit.
make -C /lib/modules/6.1.21-v8+/build M=/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module clean
make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v8+'
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'
make -C /lib/modules/6.1.21-v8+/build M=/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v8+'
CC [M] /home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.o
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:59:15: error: variable 'default_card_info' has initializer but incomplete type
59 | static struct simple_card_info default_card_info = {
| ^~~~~~~~~~~~~~~~
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:60:4: error: 'struct simple_card_info' has no member named 'card'
60 | .card = "snd_rpi_i2s_card", // -> snd_soc_card.name
| ^~~~
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:60:11: warning: excess elements in struct initializer
60 | .card = "snd_rpi_i2s_card", // -> snd_soc_card.name
| ^~~~~~~~~~~~~~~~~~
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:60:11: note: (near initialization for 'default_card_info')
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:61:4: error: 'struct simple_card_info' has no member named 'name'
61 | .name = "simple-card_codec_link", // -> snd_soc_dai_link.name
| ^~~~
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:61:11: warning: excess elements in struct initializer
61 | .name = "simple-card_codec_link", // -> snd_soc_dai_link.name
| ^~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:61:11: note: (near initialization for 'default_card_info')
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:62:4: error: 'struct simple_card_info' has no member named 'codec'
62 | .codec = "snd-soc-dummy", // "dmic-codec", // -> snd_soc_dai_link.codec_name
| ^~~~~
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:62:12: warning: excess elements in struct initializer
62 | .codec = "snd-soc-dummy", // "dmic-codec", // -> snd_soc_dai_link.codec_name
| ^~~~~~~~~~~~~~~
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:62:12: note: (near initialization for 'default_card_info')
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:63:4: error: 'struct simple_card_info' has no member named 'platform'
63 | .platform = "not-set.i2s",
| ^~~~~~~~
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:63:15: warning: excess elements in struct initializer
63 | .platform = "not-set.i2s",
| ^~~~~~~~~~~~~
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:63:15: note: (near initialization for 'default_card_info')
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:64:4: error: 'struct simple_card_info' has no member named 'daifmt'
64 | .daifmt = SND_SOC_DAIFMT_I2S | SND_SOC_DAIFMT_NB_NF | SND_SOC_DAIFMT_CBS_CFS,
| ^~~~~~
In file included from ./include/sound/soc-topology.h:15,
from ./include/sound/soc-dapm.h:15,
from ./include/sound/soc.h:417,
[... 46 lines cut ...]
124 | card_info.platform = card_platform;
| ^
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:125:12: error: invalid use of undefined type 'struct simple_card_info'
125 | card_info.cpu_dai.name = card_platform;
| ^
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c: At top level:
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:41:32: error: storage size of 'card_info' isn't known
41 | static struct simple_card_info card_info;
| ^~~~~~~~~
/home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:59:32: error: storage size of 'default_card_info' isn't known
59 | static struct simple_card_info default_card_info = {
| ^~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:250: /home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.o] Error 1
make[1]: *** [Makefile:2012: /home/pi/Raspberry-Pi-Installer-Scripts/i2s_mic_module] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'
make: *** [Makefile:35: all] Error 2
make: *** No rule to make target 'snd-i2smic-rpi.ko', needed by 'install'. Stop.
DONE.

Settings take effect on next boot.
---------------
When we write arecord -l, no device is detected.

Any help will be really appreciate.
Thanks,

cjvanlissa, 2024-06-21:
I'm having the same problem as @KePere on Raspberry Pi 3B. I see:

```
RASPBERRY_PI_3B detected.

Auto load module at boot? [y/n] y

Installing...
Reading package lists...

Building dependency tree...
Reading state information...

git is already the newest version (1:2.39.2-1.1).
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
The following packages were automatically installed and are no longer required:
libraspberrypi0 libwpe-1.0-1 libwpebackend-fdo-1.0-1
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 21 not upgraded.
fatal: destination path 'Raspberry-Pi-Installer-Scripts' already exists and is not an empty directory.
make -C /lib/modules/6.6.20+rpt-rpi-v8/build M=/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module clean
make[1]: Entering directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
make[1]: Leaving directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
make -C /lib/modules/6.6.20+rpt-rpi-v8/build M=/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
CC [M]  /home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.o
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:59:15: error: variable ‘default_card_info’ has initializer but incomplete type
   59 | static struct simple_card_info default_card_info = {
      |               ^~~~~~~~~~~~~~~~
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:60:4: error: ‘struct simple_card_info’ has no member named ‘card’
   60 |   .card = "snd_rpi_i2s_card",       // -> snd_soc_card.name
      |    ^~~~
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:60:11: warning: excess elements in struct initializer
   60 |   .card = "snd_rpi_i2s_card",       // -> snd_soc_card.name
      |           ^~~~~~~~~~~~~~~~~~
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:60:11: note: (near initialization for ‘default_card_info’)
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:61:4: error: ‘struct simple_card_info’ has no member named ‘name’
   61 |   .name = "simple-card_codec_link", // -> snd_soc_dai_link.name
      |    ^~~~
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:61:11: warning: excess elements in struct initializer
   61 |   .name = "simple-card_codec_link", // -> snd_soc_dai_link.name
      |           ^~~~~~~~~~~~~~~~~~~~~~~~
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:61:11: note: (near initialization for ‘default_card_info’)
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:62:4: error: ‘struct simple_card_info’ has no member named ‘codec’
   62 |   .codec = "snd-soc-dummy",         // "dmic-codec", // -> snd_soc_dai_link.codec_name
      |    ^~~~~
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:62:12: warning: excess elements in struct initializer
   62 |   .codec = "snd-soc-dummy",         // "dmic-codec", // -> snd_soc_dai_link.codec_name
      |            ^~~~~~~~~~~~~~~
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:62:12: note: (near initialization for ‘default_card_info’)
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:63:4: error: ‘struct simple_card_info’ has no member named ‘platform’
   63 |   .platform = "not-set.i2s",
      |    ^~~~~~~~
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:63:15: warning: excess elements in struct initializer
   63 |   .platform = "not-set.i2s",
      |               ^~~~~~~~~~~~~
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:63:15: note: (near initialization for ‘default_card_info’)
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:64:4: error: ‘struct simple_card_info’ has no member named ‘daifmt’
   64 |   .daifmt = SND_SOC_DAIFMT_I2S | SND_SOC_DAIFMT_NB_NF | SND_SOC_DAIFMT_CBS_CFS,
      |    ^~~~~~
In file included from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/soc-topology.h:15,
                 from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/soc-dapm.h:15,
                 from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/soc.h:417,
                 from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/simple_card.h:12,
                 from /home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:23:
/usr/src/linux-headers-6.6.20+rpt-common-rpi/include/uapi/sound/asoc.h:150:41: warning: excess elements in struct initializer
  150 | #define SND_SOC_DAI_FORMAT_I2S          1 /* I2S mode */
      |                                         ^
/usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/soc-dai.h:27:41: note: in expansion of macro ‘SND_SOC_DAI_FORMAT_I2S’
   27 | #define SND_SOC_DAIFMT_I2S              SND_SOC_DAI_FORMAT_I2S
[... 36 lines cut ...]
  123 |   card_info = default_card_info;
      |             ^
/home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:124:12: error: invalid use of undefined type ‘struct simple_card_info’
  124 |   card_info.platform = card_platform;
      |            ^
[... cut ...]
make[3]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/scripts/Makefile.build:248: /home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:1938: /home/cjvanlissa/Raspberry-Pi-Installer-Scripts/i2s_mic_module] Error 2
make[1]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
make: *** [Makefile:35: all] Error 2
make: *** No rule to make target 'snd-i2smic-rpi.ko', needed by 'install'.  Stop.
DONE.

Settings take effect on next boot.
```

HinTak, 2024-06-21:
Fwiw, those two are not the same problems - one is against 6.1 headers, the other against 6.6 headers. Kernel sources have evolved / changed between those. Anyway, I'll unsubscribe now, since I don't want to hear from random people commenting "same problem" on "definitely not same problem" anymore.

virtualmatador, 2024-07-03:
@cjvanlissa I forked and fixed the problem: https://github.com/virtualmatador/snd-i2smic-rpi/commit/e17a66be8ed4e33d53a7fc5992d706608fd44fa9

PR #250 (2023-05-19): Fix i2smic
Fixes #247.

This checks if the user space and kernel space are mismatched (32-bit user space and 64-bit kernel) and if so prompts to add arm_64=0 to /boot/config.txt, which solves the issue.