adafruit/Raspberry-Pi-Installer-Scripts #254: Blinka installer's Python minimum version check fails when run against Python 3.11

DerVerrater, 2023-08-03:
### Script Command

```shell
~~sudo~~ python3 raspi-blinka.py
```

### Operating System

64-bit ARM Raspberry Pi OS Desktop, Bookworm 

### Hardware

Raspberry Pi 4B, 2GB

### Behavior

Executing `raspi-blinka.py`, line 38 evaluates true, a report that the Python runtime is out of date is printed, and the program exits with a failure.

### Description

The version checking code will retrieve the Python runtime's version string and extract the first pair of numbers -- as separated by the period. This result is cast into a float, and compared against the minimum version declared as `blinka_minimum_python_version = 3.7` on line 16.

For example, the version string, `"3.9.1"` will be turned into `"3.9"`, and then the number `3.9`. The comparison `3.9 > 3.7` evaluates `True` and the test passes. The script proceeds with installation. However...

Debian 12 (Bookworm) ships with Python 3.11, which breaks when fed through this check.

The string `"3.11. 2"` is truncated to `"3.11"` and then converted to float `3.11`, and this is smaller than `3.7`. The check fails, and installation is aborted.

---

The root cause is that version numbers aren't actually numbers -- they're strings! A "correct" comparison routine would do element-wise checking. Of course, it's sufficient to pretend that they're numbers and move on with life.

My cheat to get the script to work was to adjust the number to what it meant to describe: 3.11 > 3.**0**7. Write an extra zero into the constant so the 7 is in the correct place.

### Additional information

#### Getting to Debian 12 Bookworm

The system was upgraded in-place from Debian 11 Bullseye to Debian 12 Bookworm. As far as I can tell, Raspberry Pi OS is Debian for ARM devices, with a drop in apt source in `/etc/apt/sources.list.d/raspi.source` to provide the Raspberry Pi-specific components such as the `raspi-config` script and the Broadcom device tree blobs. It *should* be okay to do a major release upgrade according to the normal Debian upgrade procedures.

I haven't seen any official Raspbian or Raspberry Pi OS images that are Bookworm just yet, nor any indication that Blinka is officially supported on Python 3.11. It's possible that the Raspberry Pi repo has a `bookworm` release, but it is considered to be "testing" still, and this install should be treated as inherently broken. For that reason, I understand if this is considered an unsupported configuration. Even still, it's a pretty silly bug, and doesn't seem like too much effort to correct.

#### Virtual environments

I chose to set up a virtual environment instead of have pip touch site package locations. The omission of `sudo` in the install command is because I logged in as root, activated the venv, and ran the installer. I had concerns over preserving the environment variables set up by the activation script of the venv, so I dodged the need for `sudo` instead of fighting with it.

araujo-fernando, 2023-09-11:
A possible correction is replacing current check_blinka_python_version method with:

```
def check_blinka_python_version():
    """
    Check the Python 3 version for Blinka (which may be a later version than we're running this script with)
    """
    print("Making sure the required version of Python is installed")
    current = get_python3_version()
    current_major, current_minor = str(current).split(".")[0:2]
    required_major, required_minor = str(blinka_minimum_python_version).split(".")[0:2]

    if current_major >= required_major and current_minor >= required_minor:
        return

    shell.bail("Blinka requires a minimum of Python version {} to install, current one is {}. Please update your OS!".format(blinka_minimum_python_version, current))
```

makermelissa, 2023-09-11:
Oh nice catch! Yeah, versioning comparison is a bit more complex.

PR #258 (2023-09-13): Update python version checking for versions > 3.9
Fixes #254.

@ladyada
closed 2023-09-13 by makermelissa