adafruit/Raspberry-Pi-Installer-Scripts #264: Script adafruit-pitft.sh fails on Raspberry Pi OS Lite (bookworm)

project-owner, 2023-10-18:
I'm trying to configure PiTFT 3.5" resistive touchscreen (480x320):
```
wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/converted_shell_scripts/adafruit-pitft.sh
chmod +x adafruit-pitft.sh
sudo ./adafruit-pitft.sh
```
error in the end:
```
[PITFT] Installing Python libraries & Software...
Installing Pre-requisite Software...This may take a few minutes!
E: Package 'python-dev' has no installation candidate
E: Package 'python-pip' has no installation candidate
E: Package 'python-smbus' has no installation candidate
E: Package 'python-spidev' has no installation candidate
WARNING : Apt failed to install software!
```
Is there any way to make it working?
Thanks!

makermelissa, 2023-10-18:
The script hasn't been updated to the latest OS yet. Maybe try changing those start with `python3` instead of `python` and try running it?

makermelissa, 2023-10-18:
Apparently there are a lot of changes that will need to be made. For now, I suggest using Bullseye. However, I'll leave the issue open.

project-owner, 2023-10-18:
Do you plan to make those changes? If so, any ETA? Thanks!
I'm using Buster right now as Bullseye has its own issues. I hoped to jump directly to the Bookworm version.

makermelissa, 2023-10-18:
I do and am working on them now.

javiercarrascocruz, 2023-10-19:
I have a similar issue when following the [1.3" 240x240 Kernel Module Install](https://learn.adafruit.com/adafruit-mini-pitft-135x240-color-tft-add-on-for-raspberry-pi/1-3-240x240-kernel-module-install). 
Running sudo python3 adafruit-pitft.py --display=st7789_240x240 --rotation=0 --install-type=console fails because python3 is no longer allowed to install packages and the famous "his environment is externally managed" error is prompted. Actually the same error occurs when installing the first dependencies at the beginning of the tutorial, but they can be installed with --break-system-packages, which does not work with the adafrit-pitft.py script.

caternuson, 2023-10-19:
Related:
https://forums.adafruit.com/viewtopic.php?p=989745

makermelissa, 2023-10-20:
I updated the python script and tested on a Raspberry Pi 5. Please don't use the shell scripts in the converted folder. Those are very out of date.

project-owner, 2023-10-20:
Great! Which script should I use? Thanks!

makermelissa, 2023-10-20:
Once #265 is merged, you should be able to follow the guide: https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/easy-install-2.

project-owner, 2023-10-20:
OK, I'll wait for the merge. I hope it's not required to run it in venv.

makermelissa, 2023-10-21:
It gives all sorts of warnings if you don't, but it is possible to run outside the venv.

javiercarrascocruz, 2023-10-21:
does it apply for my issue too? I suppose so, and therefore I would like to know how to run the  "sudo python3 adafruit-pitft.py --display=st7789_240x240 --rotation=0 --install-type=console" after the fix, once I have installed the dependencies in venv ( adafruit-python-shell and click)

javiercarrascocruz, 2023-10-21:
it still does not work for the miniPiTFT 1.3". I followed the "Easy Install" tutorial and I started the script as described:

sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=st7789_240x240 --rotation=0 --install-type=console

The script runs and I see the following at the end:
PITFT fb_st7789v.ko: 3,916 B / 16.0 KiB = 0.239     <------------ "PITFT" text is red instead of green
PITFT Updating /boot/config.txt...
Already have adafruit-pitft-helper section in /boot/config.txt.
Removing old section..
PITFT Updating console to PiTFT...
Remove fbcp from /etc/rc.local, if it's there...
PITFT Failed to disable unit: Unit file fbcp.service does not exist.     <------- "PITFTF" is red again
Configuring boot/config.txt for default HDMI
Set up main console turn on
/boot/cmdline.txt already updated
Turning off console blanking
Setting raspi-config to boot to console w/o login...
PITFT Success!

Settings take effect on next boot.

REBOOT NOW? [Y/n]

After rebooting there is nothing on the miniPiTFT, just the backlight on like before.

makermelissa, 2023-10-22:
Reopening since it doesn't appear to be fixed.

project-owner, 2023-10-22:
In my case it works to a certain extent after following the above-mentioned guide. The 'fbi' utility installed separately (sudo apt-get install fbi) can display images on 3.5" PiTFT display now:
```
/usr/bin/fbi -d /dev/fb0 --noverbose -a PATH_TO_PNG_IMAGE
```
But my Pygame 2 program cannot do that. That's probably the Pygame 2 issue. As some Adafruit software also leverages the Pygame library, it would be helpful to know if actaully works or not with PiTFT touchscreen. Also asked here: https://forums.adafruit.com/viewtopic.php?t=205374

Thank you!

makermelissa, 2023-10-24:
Ok, with further testing I found it works on the lite version, but just shows a black screen with the desktop install.

makermelissa, 2023-10-24:
I am going to close this because it works for lite and open a new issue for the full version. It appears they removed the FKMS driver, which makes this issue much bigger, albeit one I had done some work on previously.

PR #265 (2023-10-21): Update pitft to run in venv
Fixes #264. Tested on Raspberry Pi 5.

cc @ladyada
closed 2023-10-24 by makermelissa