adafruit/Raspberry-Pi-Installer-Scripts #273: Document where virtualenvs might come into play

cmdcolin, 2023-11-13:
I am a python newby and I do not really fully understand virtual envs (frankly, they confuse the heck out of me)

But they do come up trying to run some commands

For example, there are these commands in the readme

https://github.com/adafruit/Raspberry-Pi-Installer-Scripts#prepare-your-system

```
sudo apt-get install python3-pip
sudo pip3 install --upgrade click
sudo pip3 install --upgrade setuptools
sudo pip3 install --upgrade adafruit-python-shell
```

but the sudo pip3 install gives "error: externally-managed-environment"...to install python packages system-wide try etc etc etc

I am trying to juggle "should i be using the virtual env or try some other method" in my head while doing the steps from https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi#enable-i2c-and-spi-3-5 and i managed to do it but then after reboot i lost my virtual env, and then re-loading the virtual env didnt work (i can add more info if interested about this "didn't work" aspect)

thanks for all the work here!

cmdcolin, 2023-11-13:
if for example virtual envs are not recommended and the best option is to use the --break-system-packages flag that is ok too, I managed to get it working by using the --break-system-packages flag (what a scary name to try to make you use virtualenvs!)

makermelissa, 2023-12-26:
A guide covering this is now available here: https://learn.adafruit.com/python-virtual-environment-usage-on-raspberry-pi.

closed 2023-12-26 by makermelissa

cmdcolin, 2023-12-27:
thank you! looks great!