adafruit/Raspberry-Pi-Installer-Scripts #274: I2S Mic may be out of date on current Raspbian

Bradams2212, 2023-11-20:
Been trying to setup a pair of MEMS mics on my Raspberry Pi 400 using tutorials that make use of your setup script. Although it worked in the past, trying now it will run fine but does not show up as a recording device. I've asked around and they say its probably just out of date for current raspbian. If this could be updated to current or an alternative method given, that would be greatly appreciated.

makermelissa, 2023-12-26:
It looks like Bookworm made some changes and removed some sysfs stuff, which broke the i2smic stuff.

LostNulide, 2024-04-15:
I found out, that the i2s device is disabled by default in Bookworm. If you enable it in `/boot/firmware/config.` with the following line
`dtparam=i2s=on`
it should work again. At least it does on my system Raspberry Pi 4 with the latest Bookworm.

commit 3942f2cb (2024-04-15): Merge pull request #303 from LostNulide/patch-1

Fix #274
commit b5c25be3 (2024-04-15): Fix #274

This Commit depends on the Merge-Request #23 for the Adafruit_Python_Shell https://github.com/adafruit/Adafruit_Python_Shell/pull/23
closed 2024-04-15 by makermelissa