adafruit/Raspberry-Pi-Installer-Scripts #276: Raspberry Pi 5 Support

danic85, 2023-12-13:
### Script Command

```shell
cd ~
sudo pip3 install --upgrade adafruit-python-shell
wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/i2smic.py
sudo python3 i2smic.py
```

### Operating System

64 bit Raspberry Pi OS

### Hardware

Raspberry Pi 5

### Behavior

Board is rejected as it is not explicitly accepted in the installer script. Once added alongside the Pi 4 boards, this is the error:
```

Cloning into 'Raspberry-Pi-Installer-Scripts'...
make -C /lib/modules/6.1.58-v8_16k+/build M=/home/archie/Raspberry-Pi-Installer-Scripts/i2s_mic_module clean
make[1]: *** /lib/modules/6.1.58-v8_16k+/build: No such file or directory.  Stop.
make: *** [Makefile:38: clean] Error 2
make -C /lib/modules/6.1.58-v8_16k+/build M=/home/archie/Raspberry-Pi-Installer-Scripts/i2s_mic_module modules
make[1]: *** /lib/modules/6.1.58-v8_16k+/build: No such file or directory.  Stop.
make: *** [Makefile:35: all] Error 2
make: *** No rule to make target 'snd-i2smic-rpi.ko', needed by 'install'.  Stop.
DONE.
```

### Description

Following (failed) install, the mic is not available with `arecord -l`.

makermelissa, 2023-12-26:
I was able to get it to compile fine, but it still doesn't function. Looking deeper into this, the values in this section (https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/30b4c740b9c96fd8b6853ce64cff2f6d94438c20/i2s_mic_module/snd-i2smic-rpi.c#L99-L114) were originally obtained by running:
`sudo cat /sys/kernel/debug/asoc/dais`

However, for the Pi 5, only `snd-soc-dummy-dai` is returned, so it does not appear that this will run on the Pi 5 without some significant modifications.

torimos, 2024-01-19:
> I was able to get it to compile fine, but it still doesn't function. Looking deeper into this, the values in this section ( [...]

I did try to update the i2smic.py with PI5 support:
```
 elif pi_model in ("RASPBERRY_PI_5"):
        pimodel_select = 3
```

And [snd-i2smic-rpi.c](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/30b4c740b9c96fd8b6853ce64cff2f6d94438c20/i2s_mic_module/snd-i2smic-rpi.c) with according initialization:
```
  // Set platform
  switch (rpi_platform_generation) {
    case 0:
      // Pi Zero
      card_platform = "20203000.i2s";
      break;
    case 1:
      // Pi 2 and 3
      card_platform = "3f203000.i2s";
      break;
    case 2:
      // Pi 4
      card_platform = "fe203000.i2s";
      break;
    case 3:
    default:
      // Pi 5
      card_platform = "1f000a4000.i2s";
      dmaengine = "snd_pcm_dmaengine";
  }
```
The 1f000a4000.i2s I was able to get using: 
``` 
sudo cat /sys/kernel/debug/asoc/dais 
```
before running the i2smic.py script

But after making and installing module I got: 
[    4.237997] snd_i2smic_rpi: loading out-of-tree module taints kernel.
[    4.245310] snd-i2smic-rpi: Version 0.1.0
[    4.245315] snd-i2smic-rpi: Setting platform to 1f000a4000.i2s
...
[    4.305790] request module load 'snd_pcm_dmaengine': 0
[    4.311105] register platform device 'asoc-simple-card': 0

I also tried default bcm2708-dmaengine with same result

arecord -l
Does not return any recording devices

Not sure where to look next and why registration does not workl

makermelissa, 2024-01-19:
It likely doesn't work because they changed the hardware quite a bit. Also, when I run the same command on my Pi 5, I get:
```
i2s-hifi
107c706400.hdmi
i2s-hifi
107c701400.hdmi
snd-soc-dummy-dai
```

This is more than before, but still not seeing a consistent value.

torimos, 2024-01-19:
yes, it looks like RPI team is currently focused on core0 (main core) and near future there will be not much news in regards to RP1 (core1) used to communicate with I2S peripheral 
More details is here: https://forums.raspberrypi.com/viewtopic.php?t=363644

jerbzz, 2024-07-01:
Hi @makermelissa - trying to get this to run on a Pi Zero 2 W running Bookworm. May I ask what your solution was to get it to compile?

<details>
 <summary>Compiler output with a whole litany of chaos</summary>
```
make -C /lib/modules/6.6.20+rpt-rpi-v8/build M=/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module clean
make[1]: Entering directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
make[1]: Leaving directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
make -C /lib/modules/6.6.20+rpt-rpi-v8/build M=/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
CC [M]  /home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.o
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:59:15: error: variable ‘default_card_info’ has initializer but incomplete type
   59 | static struct simple_card_info default_card_info = {
      |               ^~~~~~~~~~~~~~~~
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:60:4: error: ‘struct simple_card_info’ has no member named ‘card’
   60 |   .card = "snd_rpi_i2s_card",       // -> snd_soc_card.name
      |    ^~~~
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:60:11: warning: excess elements in struct initializer
   60 |   .card = "snd_rpi_i2s_card",       // -> snd_soc_card.name
      |           ^~~~~~~~~~~~~~~~~~
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:60:11: note: (near initialization for ‘default_card_info’)
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:61:4: error: ‘struct simple_card_info’ has no member named ‘name’
   61 |   .name = "simple-card_codec_link", // -> snd_soc_dai_link.name
      |    ^~~~
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:61:11: warning: excess elements in struct initializer
   61 |   .name = "simple-card_codec_link", // -> snd_soc_dai_link.name
      |           ^~~~~~~~~~~~~~~~~~~~~~~~
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:61:11: note: (near initialization for ‘default_card_info’)
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:62:4: error: ‘struct simple_card_info’ has no member named ‘codec’
   62 |   .codec = "snd-soc-dummy",         // "dmic-codec", // -> snd_soc_dai_link.codec_name
      |    ^~~~~
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:62:12: warning: excess elements in struct initializer
   62 |   .codec = "snd-soc-dummy",         // "dmic-codec", // -> snd_soc_dai_link.codec_name
      |            ^~~~~~~~~~~~~~~
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:62:12: note: (near initialization for ‘default_card_info’)
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:63:4: error: ‘struct simple_card_info’ has no member named ‘platform’
   63 |   .platform = "not-set.i2s",
      |    ^~~~~~~~
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:63:15: warning: excess elements in struct initializer
   63 |   .platform = "not-set.i2s",
      |               ^~~~~~~~~~~~~
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:63:15: note: (near initialization for ‘default_card_info’)
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:64:4: error: ‘struct simple_card_info’ has no member named ‘daifmt’
   64 |   .daifmt = SND_SOC_DAIFMT_I2S | SND_SOC_DAIFMT_NB_NF | SND_SOC_DAIFMT_CBS_CFS,
      |    ^~~~~~
In file included from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/soc-topology.h:15,
                 from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/soc-dapm.h:15,
                 from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/soc.h:417,
                 from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/simple_card.h:12,
                 from /home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:23:
/usr/src/linux-headers-6.6.20+rpt-common-rpi/include/uapi/sound/asoc.h:150:41: warning: excess elements in struct initializer
  150 | #define SND_SOC_DAI_FORMAT_I2S          1 /* I2S mode */
      |                                         ^
/usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/soc-dai.h:27:41: note: in expansion of macro ‘SND_SOC_DAI_FORMAT_I2S’
   27 | #define SND_SOC_DAIFMT_I2S              SND_SOC_DAI_FORMAT_I2S
      |                                         ^~~~~~~~~~~~~~~~~~~~~~
/home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.c:64:13: note: in expansion of macro ‘SND_SOC_DAIFMT_I2S’
   64 |   .daifmt = SND_SOC_DAIFMT_I2S | SND_SOC_DAIFMT_NB_NF | SND_SOC_DAIFMT_CBS_CFS,
      |             ^~~~~~~~~~~~~~~~~~
/usr/src/linux-headers-6.6.20+rpt-common-rpi/include/uapi/sound/asoc.h:150:41: note: (near initialization for ‘default_card_info’)
  150 | #define SND_SOC_DAI_FORMAT_I2S          1 /* I2S mode */
      |                                         ^
/usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/soc-dai.h:27:41: note: in expansion of macro ‘SND_SOC_DAI_FORMAT_I2S’
   27 | #define SND_SOC_DAIFMT_I2S              SND_SOC_DAI_FORMAT_I2S
      |                                         ^~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
make[3]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/scripts/Makefile.build:248: /home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module/snd-i2smic-rpi.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:1938: /home/james/Raspberry-Pi-Installer-Scripts/i2s_mic_module] Error 2
make[1]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
make: *** [Makefile:35: all] Error 2
make: *** No rule to make target 'snd-i2smic-rpi.ko', needed by 'install'.  Stop.
DONE.
```
</details>

makermelissa, 2025-03-14:
I know it's been awhile. The Pi Zero 2W is the equivalent of the Pi 3, so for anyone else, I'd try that setting.