adafruit/Raspberry-Pi-Installer-Scripts #277: Raspberry Pi 5 - Import error - All GPIO related operation

the1bit, 2023-12-19:
### Script Command

```shell
import time
import adafruit_dht
import board

dht_device = adafruit_dht.DHT11(board.D4)

temperature_c = dht_device.temperature
temperature_f = temperature_c * (9 / 5) + 32
humidity = dht_device.humidity
print("Temp:{:.1f} C / {:.1f} C    Humidity: {}%".format(temperature_c, temperature_f, humidity))
```

### Operating System

6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

### Hardware

Raspberry Pi 5 Model B

### Behavior

`ImportError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/user/development/python/dht11/dht11.py", line 2, in <module>
    import adafruit_dht
  File "/home/user/development/python/dht11/env/lib/python3.11/site-packages/adafruit_dht.py", line 32, in <module>
    from digitalio import DigitalInOut, Pull, Direction
  File "/home/user/development/python/dht11/env/lib/python3.11/site-packages/digitalio.py", line 27, in <module>
    from adafruit_blinka.microcontroller.bcm2712.pin import *
  File "/home/user/development/python/dht11/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm2712/pin.py", line 5, in <module>
    from adafruit_blinka.microcontroller.generic_linux.libgpiod_pin import Pin
  File "/home/user/development/python/dht11/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 8, in <module>
    raise ImportError(
ImportError: libgpiod Python bindings not found, please install and try again! See https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/libgpiod.sh`

### Description

I want to read data from DHT11 sensor. 
(https://pimylifeup.com/raspberry-pi-dht11-sensor/)

I've installed everything on the right way, nevertheless I cannot read any data.

I cannot do any operation on GPIO. Neither sensors nor relays.

makermelissa, 2023-12-19:
Try running `sudo apt install python3-libgpiod`.

makermelissa, 2023-12-19:
Maybe try upgradin g Blinka. I just put out a release this morning that includes a libgpiod update. Try running:
`pip3 install --upgrade adafruit-blinka` in your venv.

the1bit, 2023-12-19:
Still same error. No chanes.

makermelissa, 2023-12-19:
When you installed originally, were you following this guide? https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi

the1bit, 2023-12-20:
I did it now.

When I run this: `sudo python3 raspi-blinka.py` it says:
```
raceback (most recent call last):
  File "/home/the1bit/development/python/dht11/raspi-blinka.py", line 12, in <module>
    from adafruit_shell import Shell
ModuleNotFoundError: No module named 'adafruit_shell'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/the1bit/development/python/dht11/raspi-blinka.py", line 14, in <module>
    raise RuntimeError("The library 'adafruit_shell' was not found. To install, try typing: sudo pip3 install adafruit-python-shell")
RuntimeError: The library 'adafruit_shell' was not found. To install, try typing: sudo pip3 install adafruit-python-shell
```

And when I try to execure `sudo pip3 install adafruit-python-shell` I am facing the following error:
```python
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
```

the1bit, 2023-12-20:
Both cases the virtual environment is active

the1bit, 2023-12-21:
I completely reinstalled the PI5 according to the article you sent. Error now:
[image]

Then
[image]

makermelissa, 2023-12-22:
Try using the command `sudo -E env PATH=$PATH python3 raspi-blinka.py` instead.

the1bit, 2023-12-27:
Please don't close. It's xmas

the1bit, 2023-12-28:
Hello,

I run this:
```

cd ~
source env/bin/activate
pip3 install --upgrade adafruit-python-shell
wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/raspi-blinka.py
sudo -E env PATH=$PATH python3 raspi-blinka.py

```

Then It requested a restart. I did it. 
After restart:

```
cd ~
source env/bin/activate
sudo python3 raspi-blinka.py
```

Result:
```

Traceback (most recent call last):
  File "/home/the1bit/raspi-blinka.py", line 12, in <module>
    from adafruit_shell import Shell
ModuleNotFoundError: No module named 'adafruit_shell'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/the1bit/raspi-blinka.py", line 14, in <module>
    raise RuntimeError("The library 'adafruit_shell' was not found. To install, try typing: sudo pip3 install adafruit-python-shell")
RuntimeError: The library 'adafruit_shell' was not found. To install, try typing: sudo pip3 install adafruit-python-shell

```

No changes.

Where should I continue?

the1bit, 2023-12-29:
Finally, I realized it works on another PIN:
```
# Try to great a Digital input
pin = digitalio.DigitalInOut(board.D26)
print("Digital IO ok!")
```

```
Hello blinka!
Digital IO ok!
I2C ok!
SPI ok!
done!
```

Thank you for our support.

I wish you a Happy New Year. :-)

closed 2023-12-29 by the1bit

makermelissa, 2023-12-29:
Awesome. If you need the pins that have Device or Resource busy, you can find more info here: https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/spi-sensors-devices#reassigning-or-disabling-the-spi-chip-enable-lines-3097985