adafruit/Raspberry-Pi-Installer-Scripts #284: Just bought Adafruit Speaker Bonnet but does not work with Bookworm

richsad-vcs, 2024-01-02:
Hi, I saw other messages about this, but it's rather critical for me. Downgrading the Pi's OS does not appeal to me. Can anyone perhaps either accelerate a fix or give me a pep talk on how to fix myself. I presume all other i2s devices are not working either.

Worst case, I will downgrade a Pi 0w for now. But we need I2S on Bookworm please! Thx

makermelissa, 2024-01-03:
The speaker bonnet should work fine in Bookworm. I've tested on both lite and desktop on the Pi 5 with the latest Raspberry Pi OS. Which board are you using?

richsad-vcs, 2024-01-03:
RPi3b+ but can try on a 4b+. I tried to Pimoroni scripts too. Nothing. I see a lot of other complaints it doesn't work with bookworm.

I'll try again. I even modified script to allow "bookworm" as an allowed distro. No joy.

Regards,

Rich Sadowsky

Chief Technology Officer

OFFICE: +1.832.497.5607

EMAIL: ***@***.******@***.***>

[full-color-logo_TES cropped 750x]<https://www.vikingcold.com/>

                                               [linkedinsquare] <https://www.linkedin.com/company/viking-cold-solutions/>  [facebook square] <https://www.facebook.com/vikingcold/>  [twitter square] <https://twitter.com/VikingColdSolut>

This message may contain confidential information that is proprietary to Viking Cold Solutions, Inc. and is intended only for the use of the parties to whom it is addressed. If you are not an intended recipient, please note that any disclosure, copying, distribution or use of any information in this message is prohibited. If you have received this message in error, please alert the sender by replying to this message and follow with its deletion.  Thank you for your cooperation.

makermelissa, 2024-01-03:
I just updated the i2samp.sh script a week ago, so hopefully those are old complaints.

makermelissa, 2024-01-03:
I'll test it out on a Pi 3 to make sure it works there as well.

makermelissa, 2024-01-04:
On the Pi 3, it did not work when the /dev/zero service was enabled. Disabling it by running `sudo systemctl disable aplay` and then rebooting enabled sound.

richsad-vcs, 2024-01-04:
Thanks. I will try that. Would it likely be the same for the 4b+?

Regards,

Rich Sadowsky

Chief Technology Officer

OFFICE: +1.832.497.5607

EMAIL: ***@***.******@***.***>

[full-color-logo_TES cropped 750x]<https://www.vikingcold.com/>

                                               [linkedinsquare] <https://www.linkedin.com/company/viking-cold-solutions/>  [facebook square] <https://www.facebook.com/vikingcold/>  [twitter square] <https://twitter.com/VikingColdSolut>

This message may contain confidential information that is proprietary to Viking Cold Solutions, Inc. and is intended only for the use of the parties to whom it is addressed. If you are not an intended recipient, please note that any disclosure, copying, distribution or use of any information in this message is prohibited. If you have received this message in error, please alert the sender by replying to this message and follow with its deletion.  Thank you for your cooperation.

makermelissa, 2024-01-05:
> Would it likely be the same for the 4b+?

Possibly. I think I may have previously run into the same issue on a Pi 4 and disabling that helped. I tested both with it enabled/disable on a Pi 5 and it was fine.

makermelissa, 2025-03-18:
Closing as I think this is not an issue anymore.

closed 2025-03-18 by makermelissa