adafruit/Raspberry-Pi-Installer-Scripts #287: I2S audio device not visible after installing kernel module

nightduck, 2024-01-08:
### Script Command

```shell
arecord -l
```

### Operating System

64-bit Raspberry Pi OS

    pi@raspberrypi:~ $ uname -a
    Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
    pi@raspberrypi:~ $ lsb_release -a
    No LSB modules are available.
    Distributor ID:	Debian
    Description:	Debian GNU/Linux 12 (bookworm)
    Release:	12
    Codename:	bookworm

### Hardware

Raspberry Pi 3B
Raspberry Pi 3B+

### Behavior

After running the i2smic.py script and rebooting, I can confirm that the kernel module is loaded

    pi@raspberrypi:~ $ lsmod | grep i2s
    snd_i2smic_rpi         16384  0

However, the input audio device is not visible

    pi@raspberrypi:~ $ arecord -l
    **** List of CAPTURE Hardware Devices ****

makermelissa, 2024-01-09:
Which device are you having an issue with? Which script did you use? `arecord` is not one of our scripts.

makermelissa, 2024-01-09:
Never mind. This is a duplicate of #274.

closed 2024-01-09 by makermelissa