adafruit/Raspberry-Pi-Installer-Scripts #301: i2samp.sh script "does not work on this version of raspbian" on Bookworm

Jazoolee, 2024-04-08:
### Script Command

```shell
curl -sS https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/main/i2samp.sh | bash
```

### Operating System

32-Bit Raspbian Bookworm

### Hardware

Pi 4B+

### Behavior

Does not work on this version of Raspbian

makermelissa, 2024-04-09:
What happens if you run the following commands first?
```
sudo apt update
sudo apt -y upgrade
```
After running that (and rebooting if necessary), what are the results of the following command:
```
uname -a
```
Thanks.

makermelissa, 2024-04-09:
I think this is either a duplicate of I think it may be a duplicate of #247 (because you are using 32-bit) or it's an issue with the  latest kernel, which has had some audio issues as well.

makermelissa, 2024-04-12:
Tested successfully on the 64-bit version of Pi. We may need to add a check to the script to see if the user has a mismatch between the kernel and userspace, which can cause issues. This was added to some of the converted python scripts, but i2samp wasn't converted yet, so the check wasn't added.

makermelissa, 2024-04-12:
Even better would be to update the script so that it makes use of built-in drivers so that compilation isn't necessary in the first place.