adafruit/Raspberry-Pi-Installer-Scripts #302: Giving up - Adafruit MiniPiTFT 1.3" 240x240 mirror mode

ianwilliams7158, 2024-04-12:
### Script Command

```shell
sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=st7789_240x240 --rotation=0 --install-type=mirror
```

### Operating System

Raspberry Pi OS, Legacy, 32-bit, Full version, 2024-03-12

### Hardware

Raspberry Pi Zero W v1.1
Adafruit MiniPiTFT 1.3" 240x240 display

### Behavior

Blank screen, backlight on

### Description

I've been struggling with this for days, am now giving up.  Shame, as I invested money in these screens for a project...

Hardware:

Adafruit MiniPiTFT 1.3" 240x240 - https://www.adafruit.com/product/4484
Raspberry Pi Zero W v1.1
Sandisk Ultra 16GB micro sd card

Software:

Raspberry Pi Imager v1.8.5
Raspberry Pi OS (latest version offered)

Customised settings:

set hostname - yes
set username and password - yes - pi and raspberry respectively
configure wireless lan - yes - details as appropriate (confirmed correct and working)
set locale - europe/london - GB
services enable SSH - yes with password authentication

Apply customisation? Yes
Warning confirmation - Yes

Write OS to sd card and eject when complete.

Put SD card into Pi Zero W and add power

What a few minutes to finish booting then open Putty and connect to Pi

Enter username and password

Following instructions on this page: https://learn.adafruit.com/adafruit-mini-pitft-135x240-color-tft-add-on-for-raspberry-pi/1-3-240x240-kernel-module-install

```
sudo apt update -y
sudo apt-get update -y
sudo apt-get upgrade -y
```

Setup Virtual Environment

```
sudo apt install python3.11-venv
python -m venv env --system-site-packages
```

Get this error message:

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python3.11-venv
E: Couldn't find any package by glob 'python3.11-venv'
```

Carrying on anyway...

`source env/bin/activate`

This is the result of that command in terminal:

`(env) pi@webcam1:~ $`

now reboot as required

`sudo shutdown -h now`

Remove power cable and attach screen to the far left of the gpio pins, aligning as specified on the instructions page.

Reattach power cable, wait a few minutes for boot to complete, restart Putty and re-login.  At this stage, the Pi is powered up, the screen backlight is on, but there is nothing on the screen yet.

At terminal:

```
cd ~
pip3 install --upgrade adafruit-python-shell click
sudo apt-get install -y git
git clone https://github.com/adafruit/Raspberry-Pi-Installer-Scripts.git
cd Raspberry-Pi-Installer-Scripts
sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=st7789_240x240 --rotation=0 --install-type=mirror
```

This goes through the various stages and ends with:

```
Settings take effect on next boot.

REBOOT NOW? [Y/n]
```

I choose Yes and wait for the system to reboot.  Took nearly 4 hours before Putty would allow me to log in successfully.

Still nothing displayed on the screen.  Looking at the tutorial from Adafruit, there is still more to be done.

`sudo nano rgb_display_minipitfttest.py`

Paste in the contents of the tutorial file, CTRL+X, Y, Enter

Run with command:

`python3 rgb_display_minipitfttest.py`

Get this error message:

```
Traceback (most recent call last):
  File "/home/pi/rgb_display_minipitfttest.py", line 4, in <module>
    import digitalio
ModuleNotFoundError: No module named 'digitalio'
```

Pressing buttons does nothing

Googling "linux digitalio module" tells me to install Blinka.  Found it on Adafruit here - https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi

```
cd ~
pip3 install --upgrade adafruit-python-shell
wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/raspi-blinka.py
sudo -E env PATH=$PATH python3 raspi-blinka.py
```

Completed successfully, requires reboot

`sudo reboot`

When rebooted, go back to the original tutorial and try the button script again

`python3 rgb_display_minipitfttest.py`

New error message this time:

```
Traceback (most recent call last):
  File "/home/pi/rgb_display_minipitfttest.py", line 7, in <module>
    from adafruit_rgb_display.rgb import color565
ModuleNotFoundError: No module named 'adafruit_rgb_display'
```

Google "linux module adafruit_rgb_display" - leads me to this website - https://pypi.org/project/adafruit-circuitpython-rgb-display/

`sudo pip3 install adafruit-circuitpython-rgb-display`

Install completes successfully.  Lets try the button test again:

`python3 rgb_display_minipitfttest.py`

New error message:

```
Traceback (most recent call last):
  File "/home/pi/rgb_display_minipitfttest.py", line 17, in <module>
    board.SPI(),
  File "/home/pi/.local/lib/python3.9/site-packages/board.py", line 436, in SPI
    return busio.SPI(SCLK, MOSI, MISO)
  File "/home/pi/.local/lib/python3.9/site-packages/busio.py", line 371, in __init__
    self._spi = _SPI(portId)
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_blinka/microcontroller/generic_linux/spi.py", line 25, in __init__
    self._spi = spi.SPI(device=(portid, 0))
  File "/home/pi/.local/lib/python3.9/site-packages/Adafruit_PureIO/spi.py", line 149, in __init__
    raise IOError(f"{device} does not exist")
OSError: /dev/spidev0.0 does not exist
```

This one I didn't understand, but ChatGPT to the rescue:

```
File "/home/pi/.local/lib/python3.9/site-packages/Adafruit_PureIO/spi.py", line 149, in __init__
    raise IOError(f"{device} does not exist")
OSError: /dev/spidev0.0 does not exist
```

This is looking for /dev/spidev0.0 in spi.py, but the device name in my instance is /dev/spidev0.1.  I overwrote the device name in spi.py and re-ran the button test.

It worked.  Good.  Moving on to the next part of the original tutorial.

`sudo nano stats.py`

Paste in the code from the tutorial

`python3 stats.py`

It works.  Stats are displaying correctly.  Back to the adafruit-pitft.py file.

`
cd Raspberry-Pi-Installer-Scripts
sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=st7789_240x240 --rotation=0 --install-type=mirror`

Completed successfully.  Reboot required.  Rebooted.

Blank screen, backlight on.

Mirroring the HDMI is an option for this script, but it does not work.  After investing in 6 of these screens, 3d printed enclosures and graphics, I am giving up.  I can't get it working.  I now have to redesign everything it seems.

makermelissa, 2024-04-12:
First, sorry for the frustration you are having. Are you trying to run this as a kernel mode display or through python? The first step you did was you attempted to install the kernel driver (which it sounds like it failed) and rebooted. Then you tried to set it up to run with python. On the Python Setup page is the message: "If you have already installed the kernel module, you will need to remove it by editing your /boot/firmware/config.txt file before proceeding."

With the Python method, you will only see output while a python application is outputting to the display. I'd like to get a better idea of what you are trying to achieve so I can try to see if I can replicate the issue.

ianwilliams7158, 2024-04-12:
Thanks for the reply Melissa

I am very new to Linux, and the entire ecosystem around it, so was just blindly following the setup instructions on those linked pages.  What I was ultimately trying to achieve is displaying the desktop on the screen, so I can play videos using vlc player in a model display.  VLC is being used as the museum I am creating this model for wants to use a live feed from its youtube channel on the screens.  We know the resolution won't be very good, but it's what they want, and this screen is the only one that fits in the space I have to play with.

I found another user having issues with this screen, to which LadyAda replied on github that he should be using Kernel 4.9.  I found that version and followed the same setup instructions, but still couldn't get the desktop showing on the display.  I subsequently tried with every version of Rasperry Pi OS since then, and still no joy.

For the final try, one of about a dozen on the latest version of Raspberry Pi OS, I documented everything I did, as I did it.  That file was copy/pasted above to create the original thread.  If something isn't listed above, I didn't do it.

As I've never had a problem with Adafruit products in the past, I just assumed this would be the same so went ahead and purchased everything I needed, instead of just one for testing purposes first.

makermelissa, 2024-04-12:
Ok, thanks. I've been testing these displays and it should work on the latest bookworm release with kernel 6.6. Kernel 4.9 is actually pretty old from a couple years ago, so I'm not sure if you were referencing some older suggestions for if she told you this recently. There was another issue a while back where the screen wasn't updating and it looked like static and I believe the fix for that was to use an older kernel, but that issue eventually got fixed.

I'm almost done with some guide updates and can give it a try myself in a bit. If you'd like to try again to get it working sooner, I would suggest giving it another go with the latest 64-bit version of Raspberry Pi OS. It should be working, but it's totally possible that something recently broke (happens all the time with Pi updates) and I'd like to get it working whether that means updating the setup script or guide.

ianwilliams7158, 2024-04-12:
The 32-bit version was all that was being offered by Raspbery Pi Imager, so I am manually downloading the 64-bit version to try.  I will document the process again, so, in the case it doesn't work, we can possibly work out ether what I'm doing wrong, or what is going wrong with the install.

ianwilliams7158, 2024-04-12:
That's the one I'm using too.

makermelissa, 2024-04-12:
Oh, ok. I see. 32-bit is only offered for Pi Zero W. That may be what's causing the issues.

makermelissa, 2024-04-12:
So if the 64-bit version refuses to run on the Zero W for you, then next step will be for me to load it up and see what needs to change to get this working.

makermelissa, 2024-04-12:
We'll figure out a way to get this working one way or another. The Pi Zero W is too common to not get this working.

ianwilliams7158, 2024-04-12:
Just finished downloading and burning the 64bit version to an SD card.  About to start testing.

ianwilliams7158, 2024-04-12:
I'm getting the 7 blink error, indicating the kernel is not found.  I downloaded the image directly from the Raspberry Pi website.  Will try again, but I'm not sure if it will work on the Zero W v1.1.

ianwilliams7158, 2024-04-12:
Someone on the forums is saying the Zero W v1.1 is a 32bit system. 

https://forums.raspberrypi.com/viewtopic.php?t=348482

ianwilliams7158, 2024-04-12:
Currently downloading

Raspberry Pi OS with desktop and recommended software
Release date: March 15th 2024
System: 32-bit
Kernel version: 6.6
Debian version: 12 (bookworm)
Size: 2,678MB

makermelissa, 2024-04-12:
> Someone on the forums is saying the Zero W v1.1 is a 32bit system.
> 
> https://forums.raspberrypi.com/viewtopic.php?t=348482

I suspected as much. The display used to run on 32-bit raspberry pi OS, so it should just be getting the script to put the correct settings in the config. I'll need to experiment to see what settings work.

ianwilliams7158, 2024-04-12:
Running through the setup in the original post.

Setup virtual environment - no error messages this time.

Finished install with no errors.

Rebooting.

Blank screen - backlight on.

makermelissa, 2024-04-12:
> Running through the setup in the original post.
> 
> Setup virtual environment - no error messages this time.
> 
> Finished install with no errors.
> 
> Rebooting.
> 
> Blank screen - backlight on.

Can you post the contents of your `/boot/firmware/config.txt`?

ianwilliams7158, 2024-04-12:
Rebooted again.

```
cd Raspberry-Pi-Installer-Scripts
sudo python3 adafruit-pitft.py
option 5
option4

Reboot
```

Bizarrely I then saw the final few lines of the script on the display

```
REBOOT NOW? [Y/n] y
Reboot started...

Broadcast message from root@webcam1 on pts/1 (Fri 2024-04-12 21:31:25 BST):

The system will reboot now!
```

Rebooted.

Back to blank screen with backlight on.

ianwilliams7158, 2024-04-12:
> > Running through the setup in the original post.
> > Setup virtual environment - no error messages this time.
> > Finished install with no errors.
> > Rebooting.
> > Blank screen - backlight on.
> 
> Can you post the contents of your `/boot/firmware/config.txt`?

`  GNU nano 7.2                /boot/firmware/config.txt
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

hdmi_group=2
hdmi_mode=87
hdmi_cvt=480 480 60 1 0 0 0
enable_uart=1

# --- added by adafruit-pitft-helper Fri Apr 12 21:29:14 2024 ---
[all]
hdmi_force_hotplug=1
dtparam=spi=on
dtparam=i2c1=on
dtparam=i2c_arm=on
dtoverlay=mipi-dbi-spi,spi0-0,speed=40000000
dtparam=compatible=adafruit_st7789_drm\0panel-mipi-dbi-spi
dtparam=width=240,height=240
dtparam=dc-gpio=25,backlight-gpio=22
# --- end adafruit-pitft-helper Fri Apr 12 21:29:14 2024 ---

`

makermelissa, 2024-04-12:
Thanks, that's helpful. It gives me a nice starting point. I'll see if I can get it working with that driver (probably Monday) and if needed, we can bring back the old driver which we were compiling as part of the install.

ianwilliams7158, 2024-04-12:
Thank you for your help, I honestly really appreciate it.

Ian

makermelissa, 2024-04-15:
You're welcome. I'm just wrapping up a guide update and will take a look at this next.

makermelissa, 2024-04-15:
Ok, I got it working. Here's what I did:
1. I installed for Raspberry Pi Zero and selected the default Raspberry Pi OS (Legacy, 32-bit)
2. Went through the setup according to https://learn.adafruit.com/adafruit-mini-pitft-135x240-color-tft-add-on-for-raspberry-pi/1-3-240x240-kernel-module-install
3. On the final script command, I changed "console" to "mirror" so it was `sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=st7789_240x240 --rotation=0 --install-type=mirror`
4. I edited my `/boot/config.txt` (firmware folder does not exist as this is bullseye) and found the following lines:
```
# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
max_framebuffers=2
```
5.  I removed the # from the middle line and rebooted and it's working.

Note: I couldn't find my mini pitft and used a 1.3" PiTFT bonnet for this instead, which required me to change the backlight gpio, but otherwise everything else is the same.

makermelissa, 2024-04-15:
I'll try and get the script updated so it doesn't comment out the driver for bullseye.

ianwilliams7158, 2024-04-16:
Step 1, 2 and 3 all good.

Step 4 boot/config.txt:  in nano I get the following:

`  GNU nano 7.2                    /boot/config.txt
DO NOT EDIT THIS FILE

The file you are looking for has moved to /boot/firmware/config.txt
`

and in nano at boot/firmware/config.txt, the dtoverlay line (middle as in your post) didn't have a # when I opened the file.

I'm going to reflash the card again and try from a fresh install.  BRB.

ianwilliams7158, 2024-04-16:
Ok.  Reflashed an SD card, went through the install instructions again and made the appropriate changes as specified by Melissa.  It now works perfectly.  Seems like it was that pesky # ...

Thank you for all your help, I was tearing my hair out over this.

makermelissa, 2024-04-16:
Of course. :)  I'm glad that it was only a small change needed.

makermelissa, 2024-04-16:
Reopening. I'm going to let the PR close the issue once it's merged...

PR #304 (2024-04-16): Don't comment out kms driver for mipi display
Fixes #302. So the Pi Zero W uses bullseye still (bookworm is not even an option) and after testing with the mipi driver, I found the kms driver also needs to be enabled for it to work. I added a config option so that we can selectively not have it commented out per display.

 cc: @ladyada for review
closed 2024-04-16 by makermelissa