adafruit/Raspberry-Pi-Installer-Scripts #306: Adafruit Speaker Bonnet + Left channel plays on both speakers

mikeysklar, 2024-04-22:
### Script Command

```shell
speaker-test -c2

pi@zero:~ $ speaker-test -c2

speaker-test 1.2.4

Playback device is default
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 96 to 1048576
Period size range from 32 to 349526
Using max buffer size 1048576
Periods = 4
was set period_size = 262144
was set buffer_size = 1048576
 0 - Front Left
 1 - Front Right
Time per period = 11.298986
 0 - Front Left
 1 - Front Right
Time per period = 11.320644
 0 - Front Left
 1 - Front Right
Time per period = 11.304501
 0 - Front Left
 1 - Front Right
Time per period = 11.291394
 0 - Front Left
 1 - Front Right
Time per period = 11.307376
 0 - Front Left
 1 - Front Right
Time per period = 11.319341
 0 - Front Left
 1 - Front Right
Time per period = 11.304425
 0 - Front Left
 1 - Front Right
```
```

### Operating System

32-bit Bullseye Legacy
```
pi@zero:~ $ uname -a
Linux zero 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
pi@zero:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
```

### Hardware

Pi Zero W (or 2W)
I2S Speaker Bonnet (ADA 3346)

### Behavior

right speaker play correctly during test
left audio plays on both right and left

### Description

Followed the [Voice Bonnet Pi Setup](https://learn.adafruit.com/adafruit-speaker-bonnet-for-raspberry-pi/raspberry-pi-usage) and [I2S Tweaks](https://learn.adafruit.com/adafruit-speaker-bonnet-for-raspberry-pi/i2s-tweaks). Things work smoothly, just no stand-alone left channel. 

### Additional information

Based on forum user [@a_bag_of_eagles post](https://forums.adafruit.com/viewtopic.php?t=210159).

makermelissa, 2025-03-18:
Was this tested on the Lite or Desktop version. I'm assuming lite as the Pi Zero W isn't that powerful.

makermelissa, 2025-03-18:
I'm going to try the desktop as you didn't mention lite and the desktop one in the imager doesn't specify it's the desktop version.

mikeysklar, 2025-03-19:
The testing was done on Lite. 

However, this issue can be closed. 

The [core problem came down the Bonnet itself needing a resistor change](https://github.com/adafruit/Adafruit-Stereo-Speaker-Bonnet-PCB/issues/3) on the SD_MODE.

closed 2025-03-19 by makermelissa