adafruit/Raspberry-Pi-Installer-Scripts #307: PiTFT 3.5" Plus slow touch response

mikeysklar, 2024-04-25:
### Script Command

```shell
sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=35r --rotation=90 --install-type=mirror
```

### Operating System

64-bit Pi OS Bookworm Desktop

### Hardware

Pi 4B

### Behavior

2min delay with touch screen updates

### Description

Install runs smoothly. X comes up. Touch response works. There is a two minute lag from when I press a location on the screen to when it responds.

### Additional information

No errors in dmesg. Running evtest from CLI shows immediate response to screen taps:

```
Testing ... (interrupt to exit)                                                                                                                                                                                                        Event: time 1714082994.519030, type 3 (EV_ABS), code 0 (ABS_X), value 1410                                                                                                                                                             Event: time 1714082994.519030, type 3 (EV_ABS), code 1 (ABS_Y), value 636                                                                                                                                                              Event: time 1714082994.519030, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 91                                                                                                                                                       Event: time 1714082994.519030, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1                                                                                                                                                          Event: time 1714082994.519030, -------------- SYN_REPORT ------------                                                                                                                                                                  Event: time 1714082994.526706, type 3 (EV_ABS), code 0 (ABS_X), value 1424                                                                                                                                                             Event: time 1714082994.526706, type 3 (EV_ABS), code 1 (ABS_Y), value 651                                                                                                                                                              Event: time 1714082994.526706, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 80                                                                                                                                                       Event: time 1714082994.526706, -------------- SYN_REPORT ------------                                                                                                                                                                  Event: time 1714082994.843675, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0                                                                                                                                                        Event: time 1714082994.843675, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0                                                                                                                                                          Event: time 1714082994.843675, -------------- SYN_REPORT ------------                                                                                                                                                                  Event: time 1714082994.844695, type 3 (EV_ABS), code 0 (ABS_X), value 0                                                                                                                                                                Event: time 1714082994.844695, type 3 (EV_ABS), code 1 (ABS_Y), value 0                                                                                                                                                                Event: time 1714082994.844695, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1                                                                                                                                                          Event: time 1714082994.844695, -------------- SYN_REPORT ------------                                                                                                                                                                  Event: time 1714082994.909566, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0                                                                                                                                                          Event: time 1714082994.909566, -------------- SYN_REPORT ------------                                                                                                                                                                  Event: time 1714082995.771244, type 3 (EV_ABS), code 0 (ABS_X), value 2919                                                                                                                                                             Event: time 1714082995.771244, type 3 (EV_ABS), code 1 (ABS_Y), value 1710                                                                                                                                                             Event: time 1714082995.771244, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 82                                                                                                                                                       Event: time 1714082995.771244, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1                                                                                                                                                          Event: time 1714082995.771244, -------------- SYN_REPORT ------------                                                                                                                                                                  Event: time 1714082995.778845, type 3 (EV_ABS), code 0 (ABS_X), value 2976                                                                                                                                                             Event: time 1714082995.778845, type 3 (EV_ABS), code 1 (ABS_Y), value 1684                                                                                                                                                             Event: time 1714082995.778845, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 97                                                                                                                                                       Event: time 1714082995.778845, -------------- SYN_REPORT ------------                                                                                                                                                                  Event: time 1714082995.874817, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0                                                                                                                                                        Event: time 1714082995.874817, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
```

mikeysklar, 2024-04-26:
Rolled back to Bullseye Legacy 64-bit Full install and touchscreen is very responsive and usable. No issues on that legacy release. 

[image]

makermelissa, 2024-04-26:
Good to know. In both cases, it is using the touch drivers provided by Raspberry Pi. It's possible they need to be updated and a PR submitted to Raspberry Pi.

mikeysklar, 2024-04-26:
@makermelissa -

I suspect the touch driver STMPE is good as evtest is responsive. 

Somewhere along the Xwin process of translating those coordinates to icons to opening stuff a timeout must be happening. I'm looking through the log files and see another clue.

pi5 : Bookworm 12/5/23 : $HOME/.xsession

```text
II 26-04-24 12:31:01.035 - [src/view/xdg-shell.cpp:25] New xdg popup
II 26-04-24 12:31:23.363 - [src/view/xdg-shell.cpp:25] New xdg popup
EE 26-04-24 12:31:24.719 - [types/seat/wlr_seat_touch.c:303] got touch point focus for unknown touch point
```

I'm suspicious of WAYLAND and xdg interactions.

mikeysklar, 2024-04-26:
Yeah...Wayland it is...

raspi-config --> Advanced --> W1 X11 (do not use Wayfire Wayland backend)

modify **/boot/firmware/config.txt** comment out this line:
```text
#dtoverlay=vc4-kms-v3d
```

reboot...

Display has instant response time on Pi5 Bookworm / 12/5/23 release.

mikeysklar, 2024-04-29:
@ronaldw's automated change for rolling back to the Xorg server.

```
sudo sed -e '/dtoverlay=vc4/ s/^#*/#/' -i /boot/firmware/config.txt
sudo sed -i -e '/greeter-session=/ s/=.*/=pi-greeter/' /etc/lightdm/lightdm.conf
sudo sed -i -e '/user-session=/ s/=.*/=LXDE-pi-x/' /etc/lightdm/lightdm.conf
sudo sed -i -e '/autologin-session=/ s/=.*/=LXDE-pi-x/' /etc/lightdm/lightdm.conf
```

makermelissa, 2024-04-29:
Thanks. I'm working on updating the script to temporarily switch to x11 (which should provide a temporary workaround) until we can get it working in Wayland again.

makermelissa, 2024-04-29:
I'm pretty sure the 2.8" resistive touch screen (with STMPE610) uses the same driver, so will test that out as well.

makermelissa, 2024-04-30:
To automatically change to X11, you can run `sudo raspi-config nonint do_wayland W1`, which runs the raspi-config script that runs the sed commands.

makermelissa, 2024-04-30:
#309 should fix the issue.

mikeysklar, 2024-04-30:
Confirmed.

Pi5 / Bookworm latest drop + updates works smooth with X11 rollback updates added to adafruit-pitft.py today.

mikeysklar, 2024-04-30:
Pi4 / Bookworm 3/14 drop + PiTFT 3.5" Plus

Also works smoothly with adafruit-pitft.py.

Two notable differences:

1) resolution:
* Pi5 - 480x320 
* Pi4 - 720x480 (without mouse pointer)

2) mouse pointer
* Pi5 no mouse pointer, just touchpad style 
* Pi4 mouse pointer, works better with a toothpick for selection than a finger

mikeysklar, 2024-04-30:
Should the file /boot/firmware/config.txt contain the correct resolution for this display?

change this:

```
hdmi_cvt=720 480 60 1 0 0 0
```

to this:

```
hdmi_cvt=480 320 60 1 0 0 0
```

makermelissa, 2024-04-30:
I believe the reason the resolutions are different was because it wasn't being scaled in the config.txt anymore when it was updated for wayland, which ignores those parameters and the changes are performed in the wayfire.ini file instead. However, for non-wayland devices, it should be the 720x480 because the start menu can become inaccessible otherwise due to screen height.

mikeysklar, 2024-04-30:
kk. You can close this issue.

closed 2024-04-30 by makermelissa