adafruit/Raspberry-Pi-Installer-Scripts #308: 3.5" PiTFT on Pi5 Bookworm libraspberrypi-dev

mikeysklar, 2024-04-26:
### Script Command

```shell
sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=35r --rotation=90 --install-type=mirror
```

### Operating System

64-bit Pi5 Bookworm 3/15 release + latest updates (4/26/24)

### Hardware

Pi5 + 4GB

### Behavior

Bails during install on missing package:

```text
PITFT Unable to locate package libraspberrypi-dev
```

mikeysklar, 2024-04-26:
Bookworm 12/5 64-bit full release (6.1 based kernel) works fine for install and display comes up on 3.5" TFT in mirror mode on the Pi5.

Touch response is doing the same 2+ minute delay as soon on current Bookwormon Pi4.

makermelissa, 2024-04-30:
I wasn't able to reproduce the issue on a Pi 5 with latest updates.

closed 2024-04-30 by mikeysklar