adafruit/Raspberry-Pi-Installer-Scripts #310: Pi 3.5" TFT on Bookworm 3/15 effect Pi3B / Pi Zero

mikeysklar, 2024-05-03:
### Script Command

```shell
sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=35r --rotation=90 --install-type=mirror
```

### Operating System

64 and 32 bit Bullseye

### Hardware

Pi3B and Pi Zero models

### Behavior

X fails to start. Black screen with white cursor. The same setup works fine with Pi4 / Pi5.

### Description

A recent fix for the touch interface which rolls back the X-server works well on Pi4 and Pi5, but leaves Pi3B and Pi Zero models with a black screen and blinking cursor at X launch. The fix is a simple addition of 'drm' to the /boot/firmware/config.txt:

`dtoverlay=pitft35-resistive,rotate=90,speed=20000000,fps=20,drm`

I tested this on a Pi Zero, 3B, 4B and 5. 

### Additional information

I can do a PR if you like.

makermelissa, 2024-05-03:
> I can do a PR if you like.

Please do. Thanks.

mikeysklar, 2024-05-03:
[PR #311](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/pull/311) submitted

makermelissa, 2024-05-22:
> I had the same problem on a Pi 3B with Bookworm and a 2.8" TFT. Adding `,drm` worked. Is there an estimate on when mikeysklar's PR might get reviewed? Thanks!

I just merged it if you want to test it out.

mikeysklar, 2024-05-24:
Thanks.

I found that update_configtxt() was causing problems when I was adding DRM. The drm part worked, but other options were getting dropped. I had opened issue #312 around this. 

https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/issues/312

I'll try your suggested change on a few releases and submit a PR around it.

mikeysklar, 2024-05-26:
@benjaminwp18 - What Pi OS release and which display are you using?

On a Pi 5 running Bookworm I am still able to boot and have X running without the drm option in the /boot/firmware/config.txt. We had only been adding 'drm' for pre-Bookworm releases such as Bullseye and Buster where it was needed.

The script still runs 2x when passing in something like a non-default rotation value (eg. 270):

```
sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=35r --rotation=270 --install-type=mirror
```

I also am seeing a new bug based on recent Bookworm updates. The display is reversing the X, Y mapping which was not happening before so I'll open another issue and possibly PR for that.

mikeysklar, 2024-05-26:
The new display X,Y reverse mapping was from me testing different rotations. The display was not changing, but the touch mapping calibration was getting modified in the:

```
/usr/share/X11/xorg.conf.d/20-calibration.conf
```

benjaminwp18, 2024-05-28:
I'm on Pi OS Bookworm Lite with the 2.8" capacitive display and a Pi 3B. Right now I install these packages to get an X11 kiosk:

```
xserver-xorg-video-all xserver-xorg-input-all xserver-xorg-core xinit x11-xserver-utils
```

Unless the wonky X installation is at fault, it seems like my setup pretty closely matches your setup that needed `,drm` on the Pi 3B.

mikeysklar, 2024-05-29:
Thanks for this. I'll go back to a  Pi3B bookworm to test and submit a PR.

mikeysklar, 2024-05-30:
I loaded my Pi 3B with Bookworm 3/15. Running adafruit-pitft.py dropped in code with the ",drm" into /boot/firmware/config.txt

I tried both mirror and fbcp. 

```
sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=35r --rotation=90 --install-type=mirror
```

```
sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=35r --rotation=90 --install-type=fbcp
```

```
# --- added by adafruit-pitft-helper Wed May 29 19:26:34 2024 ---
[all]
hdmi_force_hotplug=1
dtparam=spi=on
dtparam=i2c1=on
dtparam=i2c_arm=on
dtoverlay=pitft35-resistive,rotate=90,speed=20000000,fps=20,drm
# --- end adafruit-pitft-helper Wed May 29 19:26:34 2024 ---
```