adafruit/Raspberry-Pi-Installer-Scripts #316: Build libgpiod fails on RP5 on bookworm Linux

rickrcomm, 2024-05-28:
### Script Command

```shell
cd ~
# sudo apt-get install python3-pip
# sudo pip3 install --upgrade adafruit-python-shell click
# wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/libgpiod.py
# sudo python3 libgpiod.py
```

### Operating System

64-bit bookworm full desktop version

### Hardware

Raspberry Pi 5 8GB

### Behavior

.......
.......
LIBGPIOD CFLAGS="-I/tmp/libgpiod.FvAb/include/ -I/tmp/libgpiod.FvAb/tests/gpiosim \
-L/tmp/libgpiod.FvAb/lib/.libs/ -L/tmp/libgpiod.FvAb/tests/gpiosim/.libs/" \
/usr/bin/python -m build
LIBGPIOD /usr/bin/python: No module named build
LIBGPIOD make[4]: *** [Makefile:689: all-local] Error 1
LIBGPIOD make[4]: Leaving directory '/tmp/libgpiod.FvAb/bindings/python'
LIBGPIOD make[3]: *** [Makefile:401: all-recursive] Error 1
LIBGPIOD make[3]: Leaving directory '/tmp/libgpiod.FvAb/bindings/python'
LIBGPIOD make[2]: *** [Makefile:398: all-recursive] Error 1
LIBGPIOD make[2]: Leaving directory '/tmp/libgpiod.FvAb/bindings'
LIBGPIOD make[1]: *** [Makefile:467: all-recursive] Error 1
LIBGPIOD make[1]: Leaving directory '/tmp/libgpiod.FvAb'
LIBGPIOD make: *** [Makefile:397: all] Error 2

### Description

I saved the whole build output and can send you if you want.

aamargulies, 2024-06-09:
Initially I got the same error as shown here.

But all that was necessary to get a pitft35-resistive display working on
a raspberry pi 5 running Debian GNU/Linux 12 (bookworm) was:

# sudo apt install python3-build
# sudo /usr/bin/python libgpiod.py

I did have to download pitft35-resistive-overlay.dts to a directory named overlays
and then run adafruit-pitft.py, but it does work now.

makermelissa, 2024-06-21:
If you need to use libgpiod on the pi, I'd recommend `pip install gpiod` for the time being.

makermelissa, 2025-08-26:
Closing as this isn't necessary.

closed 2025-08-26 by makermelissa