adafruit/Raspberry-Pi-Installer-Scripts #320: i2smic.py vs dtoverlay=googlevoicehat-soundcard

mikeysklar, 2024-07-02:
A flurry of I2S issues [popped up in the forums](https://forums.adafruit.com/viewtopic.php?t=211536) with regards to setting up the SPH0645 on Legacy Pi OS. The current guide involves compilation of modules which can result in compilation failure. 32-bit vs 64-bit mismatch and other reasons.

One user pointed out using the the googlevoicehat module just works. It is a single line change with no other configuration necessary to start recording with the [SPH0645 I2S MEMS mic](https://www.adafruit.com/product/3421).

`dtoverlay=googlevoicehat-soundcard`

I've tested this on 32-bit / 64-bit PiOS Bullseye Legacy and Bookworm. They all produce valid sound files. Pi Zero and Pi4 hardware.

[image]

Does it make sense for me to [update the guide](https://learn.adafruit.com/adafruit-i2s-mems-microphone-breakout/) and not use i2smic.py for this device?

makermelissa, 2024-07-02:
Yes, let's go ahead with that. Using built-in Device Tree Overlays is preferable from a support/future proofing standpoint.

mikeysklar, 2024-07-02:
Okay, I will update the SPH0645 guide to suggest the single line addition to config.txt. It appears the googlevoicehat module has been available in some form since 2017 so it looks like a stable option going forward.

mikeysklar, 2024-07-03:
I updated the first half of the guide page to use the one liner config.txt mod with the googlevoicehat-soundcard. 

The volume control would still be nice, but I need to verify it takes with the new driver so I've not modified that part of the page yet.

mikeysklar, 2024-07-05:
Kind of a bummer, but no volume control / mic gain with the googlevoicehat-soundcard dtoverlay. 

[image]

closed 2024-07-05 by mikeysklar