adafruit/Raspberry-Pi-Installer-Scripts #322: i2samp.py doesn't comment out dtparam=audio=on from config.txt as required

Patronics, 2024-09-09:
### Script Command

```shell
wget https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/raw/main/i2samp.py
sudo -E env PATH=$PATH python3 i2samp.py
```

### Operating System

64 bit Raspberry Pi OS (2024-07-04 release plus updates from apt update/apt upgrade). 

### Hardware

Raspberry Pi 4B

### Behavior

Error message when trying to run the installer after the reboot to test the system: 
```
XDG_RUNTIME_DIR (/run/user/1000) is not owned by us (uid 0), but by uid 1000! (This could e.g. happen if yu try to connect to a non-root PulseAudio as a root user, over the native protocol. Don't do that.)
ALSA lib pcm_dmix.c: 999: (snd_pcm_dmix_open) unable to open slave
speaker-test 1.2.8

Playback device is default
Stream parameters are 48000Hz, S16_LE, 2 channels
WAV file(s)
Playback open error: -524, Unknown error 524

ALL done!

Enjoy your new Sproductname!
REBOOT NOW? [Y/n]
```
And speaker output is nonfunctional.

also no output from `speaker-test -c2` until I manually adjust  `/boot/firmware/config.txt` to comment out `dtparam=audio=on` and reboot.

Patronics, 2024-09-10:
Additionally, the audio starting produces loud pops, despite the instructions claiming this issue has been resolved by the current installer. 

This can be worked around by starting background audio playing, such as by running ffplay on a silent file.

greenlamb, 2025-02-20:
Thank you for this bug report, I had also encountered the same issue with no output from `speaker-test -c2` until `dtparam=audio=on` was commented out.

makermelissa, 2025-08-26:
Tested recently with new changes and having the `dtparam=audio=on` did not cause issues.

closed 2025-08-26 by makermelissa