adafruit/Raspberry-Pi-Installer-Scripts #323: blinkatest.py fails on Pi 5

mikeysklar, 2024-09-19:
### Script Command

```shell
GPIO has changed on the Pi 5. Currently Bookworm 64-bit Lite fails to run blinkatest.py with the following error. 

$ python3 blinkatest.py
Hello, blinka!
Traceback (most recent call last):
  File "/home/pi/blinkatest.py", line 8, in
    pin = digitalio.DigitalInOut(board.D4)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/env/lib/python3.11/site-packages/digitalio.py", line 193, in __init__
    self.direction = Direction.INPUT
    ^^^^^^^^^^^^^^
  File "/home/pi/env/lib/python3.11/site-packages/digitalio.py", line 223, in direction
    self._pin.init(mode=Pin.IN)
  File "/home/pi/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pin.py", line 40, in init
    GPIO.setup(self.id, GPIO.IN)
RuntimeError: Cannot determine SOC peripheral base address
```

This resolves it. 

```
sudo apt remove python3-rpi.gpio
pip3 install rpi-lgpio
python3 blinkatest.py
```

```
$ python3 ./blinkatest.py
Hello, blinka!
Digital IO ok!
I2C ok!
SPI ok!
done!
```

I'll look into a PR for raspi-blinka.py used to setup the blinka env.
```

### Operating System

64-bit Lite Bookworm

### Hardware

Pi 5

### Behavior

.

### Description

.

### Additional information

Based on [forum issue](https://forums.adafruit.com/viewtopic.php?t=213519) from @bonclay.

ladyada, 2024-09-20:
ok yah can you open a PR for me to review please 🙏 and tag me in a comment

mikeysklar, 2024-09-20:
will do.

mikeysklar, 2024-09-20:
[PR# 324 ](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/pull/324) tested on Pi4 and Pi5. Getting a clean blinkatest.py after running raspi-blinka.py on both models. Running Bookworm 64-bit Lite.

closed 2024-09-26 by mikeysklar