adafruit/Raspberry-Pi-Installer-Scripts #325: adafruit-pitft.py - Red warnings if  /etc/gdm3/custom.conf does not exists.

riktam, 2024-09-22:
Running adafruit-pitft.py on Ubuntu  22.04.5 LTS 
I get the following "red lines":

```
PITFT grep: /etc/gdm3/custom.conf
PITFT : No such file or directory
PITFT sed: can't read /etc/gdm3/custom.conf: No such file or directory

```
The script does not stop and seems to finish correctly but it's an error anyway.

makermelissa, 2025-03-13:
It is probably trying to install something, which causes the error, rather than the error being caused by the script itself as there is no reference to /etc/gdm3. I think the fix is going to be to not do any compile software installs if the display doesn't need it (i.e. boards with "kernel_upgrade": False), which is why I think the install worked for you.

makermelissa, 2025-03-13:
I looked further into this and gdm3 is GNOME display manager. This script has only been tested with Wayland and X11, so maybe that has something to do with it.

riktam, 2025-03-13:
In my setup I have no display manager installed.

makermelissa, 2025-08-26:
Closing as this is only a warning and the script works.

closed 2025-08-26 by makermelissa