adafruit/Raspberry-Pi-Installer-Scripts #331: adafruit-pitft.py :: "Programmer error, unrecognized boot option"

mikeysklar, 2025-01-29:
### Script Command

```shell
sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=35r --rotation=90 --install-type=console
```

### Operating System

Pi OS Bookworm 11/19/24 64-bit Desktop

### Hardware

Raspberry Pi 4B

### Behavior

adafruit-pitft.py installer script has started bailing since a set of updated packages were released this week. The error message is:

> Programmer error, unrecognized boot option

### Description

The lines that are causing this are from the raspi-config boot options:

```
#        if not shell.run_command("sudo raspi-config nonint do_wayland 0"):
#        if not shell.run_command("sudo raspi-config nonint do_wayland 1"):
```

Commenting them out did allow my Pi4B to run through the full adafruit-pitft.py and boot into console mode. 

Open forum issue with two other uses are different PiTFTs using Pi 3B are seeing this as well. All users of the adfafruit-pitft.py will be effected with the current Bookworm updates applied.

mikeysklar, 2025-01-29:
@makermelissa

mikeysklar, 2025-01-29:
It might just be a number change to raspi-config.

eg. 0,1 becomes W1, W2, W3 based on [raspi-config asciidoc](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/configuration/raspi-config.adoc#wayland-1)

console booting:
```
$ sudo raspi-config nonint do_boot_behaviour <B1/B2/B3/B4>

B1: boot to console, requiring login

B2: boot to console, logging in automatically

B3: boot to desktop, requiring login

B4: boot to desktop, logging in automatically
```

X11 vs Wayland booting:
```
$ sudo raspi-config nonint do_wayland <W1/W2/W3>

W1: use the Openbox window manager with X11 backend

W2: use the wayfire window manager with Wayland backend

W3: use the labwc window manager with Wayland backend
```

mikeysklar, 2025-01-29:
Yeah...minor argument renumbering. 

I'll drop in a PR. There might be more to add to it as I think this update could have broken other components.

makermelissa, 2025-01-30:
This was likely caused by https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/pull/327.

mikeysklar, 2025-01-30:
Thx. Myself and another user verified #332 works, but there is some noise about various config files. I'll clean that up in a separate PR.

```
Remove fbcp from /etc/rc.local, if it's there...
PITFT Failed to disable unit: Unit file fbcp.service does not exist.
Configuring boot/config.txt for default HDMI
Restoring Wayland as default display manager...
Using Wayland instead of X11
PITFT sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
PITFT sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
PITFT sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
PITFT sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
PITFT sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
PITFT sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
```

closed 2025-01-30 by mikeysklar