adafruit/Raspberry-Pi-Installer-Scripts #334: Issues with raspi-blinka.py and Raspberry Pi 5

BlitzCityDIY, 2025-02-19:
### Script Command

```shell
sudo -E PATH=$PATH python raspi-blinka.py
```

### Operating System

64-bit Raspberry Pi OS Bullseye

### Hardware

Raspberry Pi 5

### Behavior

The script executes properly and recognizes that its on a Pi 5. When running the Blinka test script, I got the "Cannot determine SOC peripheral base address". I commented out `#dtparam=spi=on` and rebooted. I still got an error.
I referenced the Blinka manual install instructions and found the RPi.GPIO note on Pi 5 and ran `pip uninstall -y RPi.GPIO` and it uninstalled `RPi.GPIO`. Running the Blinka test script, it got further but then failed on the SPI check. I went back into config.txt and uncommented `dtparam=spi=on` and rebooted. Finally the Blinka test script ran properly.

This was confusing to navigate and I was referencing the Learn guide. This was on a fresh install made with Raspberry Pi Imager, released 2024-11-19.

### Description

When I ran the script on a fresh install on a Pi 5 in October for [this project](https://learn.adafruit.com/nfc-raspberry-pi-media-player/overview), I did not run into any of these issues.

makermelissa, 2025-03-13:
This is a known issue. Please refer to https://github.com/adafruit/Adafruit_Blinka/issues/907#issuecomment-2605477551

closed 2025-03-13 by makermelissa

makermelissa, 2025-03-14:
BTW, this should be working now because of https://github.com/adafruit/Adafruit_Blinka/pull/911