adafruit/Raspberry-Pi-Installer-Scripts #343: 2.8" capacitive touchscreen shows nothing after install script

Kheirlb, 2025-04-27:
### Script Command

```shell
sudo -E env PATH=$PATH python3 adafruit-pitft.py --display=28c --rotation=90 --install-type=console
```

### Operating System

https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2023-12-11/

### Hardware

Pi: pi400
Touchscreen: 2.8" capacitive touchscreen

### Behavior

Screen flashes white during reboot, but screen never shows anything besides black.

### Description

Some of the errors in the end of the script are related to fcpb/lightdm.
```
PITFT Updating /boot/firmware/config.txt...
PITFT Updating SysFS rules for Touchscreen...
PITFT Updating TSLib default calibration...
PITFT Updating console to PiTFT...
Remove fbcp from /etc/rc.local, if it's there...
PITFT Failed to disable unit: Unit file fbcp.service does not exist.
Configuring boot/config.txt for default HDMI
Restoring Wayland as default display manager...
Using wayland as the window manager
PITFT sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
PITFT sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
PITFT sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
PITFT sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
PITFT sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
PITFT sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
Set up main console turn on
Updating /boot/firmware/cmdline.txt
Turning off console blanking
Setting raspi-config to boot to console w/o login...
PITFT Success!

Settings take effect on next boot.

REBOOT NOW? [Y/n]
```

### Additional information

Link to discord thread https://discordapp.com/channels/327254708534116352/1362690039330635876

Kheirlb, 2025-04-27:
I've had this capacitive TFT for years, and I have never figured out why there is only this blank screen. Did I get a dud maybe? I ran through the most recent guide and easy install scripts with no luck.

[image]
[image]

makermelissa, 2025-05-20:
I'm taking a look at the script. It may be having issues with the lite version of pi and needs some updates.

makermelissa, 2025-05-20:
Please try again. I ran the script, rebooted and it works for me. Regarding the error messages, they are harmless. I just made PR #344 so the messages aren't displayed in the future.

makermelissa, 2025-07-02:
Closing due to no response. Please feel free to reopen if you test again and are still experiencing the same issue.

closed 2025-07-02 by makermelissa