adafruit/Raspberry-Pi-Installer-Scripts #350: rgb-matrix.sh: does not install compiled python binary into the system

WeegeeNumbuh1, 2025-07-15:
### Script Command

```shell
sudo bash rgb-matrix.sh
```

### Operating System

[DietPi v9.14](https://dietpi.com/docs/releases/v9_14/), 64-bit

Also tested with an existing system running Raspberry Pi OS Bookworm, 32-bit w/ desktop that had used the `rgb-matrix.sh` script from a year ago.

### Hardware

Raspberry Pi Zero 2W w/ RGB Matrix Bonnet and quality mod

### Behavior

When running the script to build the rgbmatrix library, it does not install the built binary into the system, nor does it update a previously installed version of rgbmatrix.

### Description

In the last big commit for the script https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/commit/a29743d6af6e6be0e0c19b8e4803d7f3e748b9a5 the lines involving `make install-python` were removed. Since this method is depreciated (as it calls `setup.py` directly), it was understandably removed. However, no replacement action was implemented in the script.

After running the script and rebooting following the last prompt, the expected behavior once a python interpreter is fired up is to be able to use `import rgbmatrix`. This does not occur and results in a `ModuleNotFound` error.

When running the sample python projects in the `samples` directory under the `bindings/python` directory, they work fine, thus leading me to believe the freshly built binaries were not installed globally on the system.

### Additional information

I restored the old script's functionality by adding these lines after the `make build-python USER_DEFINES="$USER_DEFINES"` line:

```
export PIP_ROOT_USER_ACTION=ignore
cd bindings/python
pip install . --use-pep517 --break-system-packages
```

Once this modified script was run, it builds the `wheel` correctly and I was able to call `import rgbmatrix` globally.

_Additional notes regarding the DietPi system:_
As the image is extremely minimal, I had to install other system packages for completeness before running this script (this also assumes `python` was installed during the initial setup):
```
sudo apt install -y make unzip build-essential
```

sperrow, 2025-07-19:
Also seeing this issue

FoamyGuy, 2025-07-24:
In #346  it mentions about this:
> Only builds Python bindings but does not install them, since the make files do not support Python venv.

Going to close this, but we can re-open if it warrants further discussion.

closed 2025-07-24 by FoamyGuy

caternuson, 2025-07-28:
This is intentional and is related to Python virtual environments. More info in this guide:
https://learn.adafruit.com/python-virtual-environment-usage-on-raspberry-pi/overview

Some info here on how to install the RGB matrix Python bindings:
https://learn.adafruit.com/adafruit-rgb-matrix-plus-real-time-clock-hat-for-raspberry-pi/install-manually#python-build-3201099

The previous guide/script was written before the change to enforcing venv usage.